ros2/geometry2 #5: ROS2 time vs TF Time

Karsten1987, 2017-03-27:
I've seen that `get_now()` only returns the system clock. Does this eventually lead to out-of-sync with `rclcpp`-Time (ROS-time, system-time, steady-time, etc.)?

tfoote, 2017-03-27:
Can you point to the line of code you're referring to?

Karsten1987, 2017-03-27:
I am referring to this: 
https://github.com/ros2/geometry2/blob/ros2/tf2/include/tf2/time.h#L52

Also the `Timepoint` is typedef'ed to `std::chrono::system_clock`: https://github.com/ros2/geometry2/blob/ros2/tf2/include/tf2/time.h#L43

tfoote, 2017-03-27:
gosh github search is terrible. 

Yes, that will need to be updated to support ros::Time once we have an implementation of rostime/clock

Like this: https://github.com/ros/geometry2/blob/439e2350ca9b2f0581a3631f0929acc2021db73e/tf2_ros/src/buffer.cpp#L73-L84

matt-attack, 2018-02-19:
Just want to give this a bump. rclcpp::Time is implemented now.

sloretz, 2018-09-21:
Fixed by ros2/geometry2#67 , closing

closed 2018-09-21 by sloretz