ros2/geometry2 #6: tf2_ros buffer interface transform() methods

dhood, 2017-03-28:
Opening issue to avoid duplicating work. Has anyone ( @clalancette) looked at porting the transform() methods in tf2_ros/buffer_interface.h yet? They're still [commented out](https://github.com/ros2/geometry2/blob/1c4d964340b4ee73eb596fe7c75e7d1d459d3ef9/tf2_ros/include/tf2_ros/buffer_interface.h#L141)

clalancette, 2017-03-28:
I have not; I did not need them for cartographer, so I didn't end up touching those.  Go ahead and work on them if you need them.

dhood, 2017-03-28:
:+1: thanks

tfoote, 2017-06-14:
Mostly resolved in #20 however the full power is not available because the templated convert(A,B) is still commented out: https://github.com/ros2/geometry2/blob/ros2/tf2/include/tf2/convert.h#L113

clalancette, 2020-03-16:
This was actually done a while ago, so closing this out.

closed 2020-03-16 by clalancette