ros2/geometry2 #8: delete ros/geometry2 ros2 branch

Karsten1987, 2017-04-04:
Now that we forked this repo and started opening PRs against this repos' ros2 branch. Can we delete the ros/geometry2 ros2 branch?

tfoote, 2017-04-04:
That's fine as long as we have all the content here.

mikaelarguedas, 2017-04-05:
Needs to make sure that all the code from geometry/ros2 is in geometry2/ros2.
Also it would be great to sync geometry2/indigo-devel with geometry/indigo-devel and rebase the ros2 branch on top of indigo-devel

clalancette, 2017-04-05:
I just took a quick look; it does look like the graphs are the same, though they have different commit hashes.  I don't recall if we ended up rebasing the ros2/geometry2 ros2 branch, but that would explain it.  So I think we are pretty good there.

I wonder if it is even worth it to sync indigo-devel; there isn't really a lot of benefit, since if we ever to want to remerge the two, it should be a simple fast-forward at that point.  I'm not sure it is worth the ongoing maintenance to continue to resync the two.

mikaelarguedas, 2017-04-05:
I thought that the decision of contributing the ROS2 agnostic changes upstream was to just have to stay in sync with upstream to have these changes. In this scenario I'm not sure what the suggested alternative is: do you mean let indigo-devel diverge on the fork and rebase the ros2 branch against upstream/indigo-devel wqhen there is a change upstream?

clalancette, 2017-04-05:
Yeah, exactly.  Though I guess at that point, it's not much more work to just push to indigo-devel in ros2/geometry2, so either way :).

Karsten1987, 2017-04-13:
what's the status on this one here?

mikaelarguedas, 2017-04-13:
well we will need to resolve conflicts at some point and the later the more time it will take so I think that this should be done soon given that there is a lot of tf2 changes coming up this cycle. If there is a strong preference to rebase against `upstream/indigo-devel` rather than an up-to-date `indigo-devel` I won't argue strongly against it (even if it seems counter-intuitive to me when I rebase locally). But my point here is that the rebase/conflict resolution should happen soon in both cases.

clalancette, 2017-04-13:
I think at this point the only thing to do to get this done is to have someone with ros/geometry2 commit rights click the button to delete the ros2 branch.

mikaelarguedas, 2017-04-13:
indeed, I'm discussing the strategy for ros2/geometry2 in general and not the deletion of the branch itself. You confirmed the code is the same on both so it's totally fine to delete it

tfoote, 2017-04-13:
Since the code is elsewhere I've deleted the branch.

closed 2017-04-13 by tfoote

Karsten1987, 2017-04-13:
thanks @tfoote for taking care of it.

mikaelarguedas, 2017-04-14:
@tfoote can you restore the branch? beta1 was using it so now people following the installation instruction cant clone their workspace (c.f https://github.com/ros2/geometry2/issues/11)

mikaelarguedas, 2017-04-14:
Update: I pushed the ros2 branch back to ros/geometry2, tagged it as release-beta1 and deleted it again this way people can vcs import our repos file from beta1