ros2/geometry2 #11: Error while checkout with git

Alex031544, 2017-04-14:
Hello,

I tryed to clone/checkout this package on different machines and received the following error:

```
ros/geometry2:
---------------------------------------------------------------
mkdir -p /tmp/ros/ros2_ws/src/ros
>> cloning sources
change path to: /tmp/ros/ros2_ws/src/ros
git clone https://github.com/ros/geometry2.git
Cloning into 'geometry2'...
remote: Counting objects: 8784, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 8784 (delta 0), reused 0 (delta 0), pack-reused 8781
Receiving objects: 100% (8784/8784), 2.04 MiB | 610.00 KiB/s, done.
Resolving deltas: 100% (5363/5363), done.
Checking connectivity... done.
>> checkout desired version
change path to: /tmp/ros/ros2_ws/src/ros/geometry2
git checkout 3a820ee7238e406282c901991186adc6558cdb5e
fatal: reference is not a tree: 3a820ee7238e406282c901991186adc6558cdb5e
error while checkout with git
```

mikaelarguedas, 2017-04-14:
Hi @Alex031544, we now rely on this fork so you chould clone github.com/ros**2**/geometry2
If you already had a checkout of ros/geometry2 in the same workspace you will want to change the remote and pull from our current development branch or just delete the folder and clone using this URL: https://github.com/ros2/geometry2

Hope this helps

Alex031544, 2017-04-14:
Hello, I commented the ros-Packages in _ros2.repos_ out. This is enough to me. Maybe you change it too...

Thanks

closed 2017-04-14 by Alex031544

mikaelarguedas, 2017-04-14:
can you point to what ros2.repos file you were using ? Thanks!

Alex031544, 2017-04-14:
I load it from here today:
```
wget https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos
```
I attach the file: [ros2.repos](https://github.com/ros2/geometry2/files/923026/ros2.repos.txt)

and these one I comment out:
```
  ros/class_loader:
    type: git
    url: https://github.com/ros/class_loader.git
    version: a7e9af992f9fb3ba6b859203a322153ee82f4de5
  ros/geometry2:
    type: git
    url: https://github.com/ros/geometry2.git
    version: 3a820ee7238e406282c901991186adc6558cdb5e
```

mikaelarguedas, 2017-04-14:
ok thanks for pointing it, that's an oversight on our side. You can keep class_loader (you will very likely be needing it tu use ROS2). We'll restore the missing branch so that the installation instructions you followed work again. Thanks for reporting it !

mikaelarguedas, 2017-04-14:
It should be fixed now, you should be able to vcs import / clone the  https://raw.githubusercontent.com/ros2/ros2/release-latest/ros2.repos file as is now