ros2/geometry2 #12: the rampant LOOPING msg

BrannonKing, 2017-04-15:
What do we need to do to get rid of this warning? It spews a lot of useless output.
```
ROS_INFO("LOOPING due to no latching at the moment\n")
```

clalancette, 2017-04-17:
My understanding is that we don't currently have latching functionality in ROS2, so we need to continually publish the transform from static_transform_broadcaster.  It looks like we do that every 5 seconds, and we print that message out at the same time.  We also don't currently have proper console logging implemented in ROS2, so there is no easy way to filter it out.  The correct way to fix it would be to fix one of those two things, but they aren't simple tasks.

In the meantime, I kind of like having the message, as it is a reminder that it is still broadcasting.  However, we could quiet it down a bit, and print less often.  Would it be better if we printed every 30 seconds (say) instead, of every 5 (we would still broadcast every 5, just print less often)?

mikaelarguedas, 2017-04-17:
Note: @dirk-thomas has been working on the console logging lately, if you want to contribute to that part, make sure to sync with him to see what can be parallelized and to be sure not to duplicate work

BrannonKing, 2017-04-17:
30 seconds would definitely be better than 5.

wjwwood, 2017-04-17:
The durability setting is like latching in ros 1. We just didn't have it when @tfoote converted tf.

BrannonKing, 2017-04-17:
Yes, I was going to say: I'm certain all our DDS middlewares support latching.

clalancette, 2017-07-14:
Hm, all of that being said, I just took a quick look at Fast-RTPS master.  From all indications it looks like Fast-RTPS doesn't support PERSISTENT durability yet, so I don't think we can reasonably change this to not loop.  Also, we don't have PERSISTENT durability plumbed into the RMW layer.  Thus, I think we need to do a few things:

- Implement durability in Fast-RTPS: I've opend an issue there eProsima/Fast-RTPS#120
- Plumb durability into rmw; we at least need ros2/rmw#113 , there is probably more
- For now, reduce the frequency we print these updates out.  I've opened PR #36 for that.

wjwwood, 2017-07-14:
We don't need persistent I think.

wjwwood, 2017-07-14:
Sorry hit send too early. We just need transient local. Which we already have exposed. 

Persistent requires a service, like a database or something, to work. We didn't have that in ros 1 and I think it's overkill for this problem.

clalancette, 2017-07-14:
So, I tried this out.  I went into https://github.com/ros2/geometry2/blob/ros2/tf2_ros/src/static_transform_broadcaster.cpp#L44 , and changed the durability setting of that custom_qos_profile to RMW_QOS_POLICY_DURABILITY_TRANSIENT_LOCAL.  I then changed the end of static_transform_broadcaster_program.cpp to:

```
  broadcaster.sendTransform(msg);
  ROS_INFO("Spinning until killed publishing %s to %s", msg.header.frame_id.c_str(), msg.child_frame_id.c_str());
  rclcpp::spin(node);
```

From what I can tell, the transform doesn't get properly published that way; at least, cartographer (which wants to consume the transform) doesn't see it.  Do I need to do something else to make TRANSIENT_LOCAL work?

dhood, 2017-07-14:
yeah, you need to "request" transient_local on the cartographer side in the subscriber's qos. there's info about this in https://github.com/ros2/ros2/wiki/About-Quality-of-Service-Settings#qos-compatibilities if you haven't seen it already (if you have, please suggest changes because you won't be the only one left asking this question!).

the challenge is that "request" is more like "require" and the subscriber will no longer be able to subscribe to volatile publishers. That's definitely missing from the wiki, different to ROS 1, and going to be an issue, especially in the case of TF where you want to subscribe to a variety of publishers, not just static transform publishers with transient local. @wjwwood and I have been discussing different options but we didn't come to a conclusion for an easy fix for the moment. I'll try to summarise next week so we can at least get the info out there.

tfoote, 2018-02-22:
This needs a decision on how to implement latching. Ticketed at ros2/ros2#464

clalancette, 2019-09-16:
As of #160 , we've removed the message completely and now use a transient local publisher.  Closing.

closed 2019-09-16 by clalancette