ros2/geometry2 #15: BufferCore confuses quaternion coordinates

BrannonKing, 2017-04-19:
There is a nasty bug in https://github.com/ros2/geometry2/blob/ros2/tf2/src/buffer_core.cpp line 143. It passes the quaternion arguments in the wrong order. It should be x, y, z, w. 

I'm not sure if the underlying Quaternion has always taken them in the order it does now, or this bug was introduced recently. The various quaternion frameworks have a variety of point orderings; we must be very careful in this regard.

The affect of this bug is that any tool that pulls transforms using `tf2_ros;:Buffer::lookupTransform` will have the wrong rotation (unless it's all zeros). Cartographer relies on this feature.

BrannonKing, 2017-04-19:
Tagging @clalancette

tfoote, 2017-04-19:
Definitely a bug https://github.com/ros2/geometry2/blob/ros2/tf2/src/buffer_core.cpp#L143 It appears I made the typo in the beginning of the ros2 porting process.

closed 2017-04-25 by clalancette