ros2/geometry2 #24: new test failure in tf2_geometry_msgs package

wjwwood, 2017-06-19:
```
-- run_test.py: invoking following command in '/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs':
 - /home/rosbuild/ci_scripts/ws/build/tf2_geometry_msgs/test_tf2_geometry_msgs --gtest_output=xml:/home/rosbuild/ci_scripts/ws/build/tf2_geometry_msgs/test_results/tf2_geometry_msgs/test_tf2_geometry_msgs.gtest.xml
[==========] Running 3 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 3 tests from TfGeometry
[ RUN      ] TfGeometry.Frame
/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:57: Failure
The difference between v_simple.pose.orientation.x and 0.0 is nan, which exceeds EPS, where
v_simple.pose.orientation.x evaluates to -nan,
0.0 evaluates to 0, and
EPS evaluates to 0.001.
/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:58: Failure
The difference between v_simple.pose.orientation.y and 0.0 is nan, which exceeds EPS, where
v_simple.pose.orientation.y evaluates to -nan,
0.0 evaluates to 0, and
EPS evaluates to 0.001.
/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:59: Failure
The difference between v_simple.pose.orientation.z and 0.0 is nan, which exceeds EPS, where
v_simple.pose.orientation.z evaluates to -nan,
0.0 evaluates to 0, and
EPS evaluates to 0.001.
/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:60: Failure
The difference between v_simple.pose.orientation.w and 1.0 is nan, which exceeds EPS, where
v_simple.pose.orientation.w evaluates to -nan,
1.0 evaluates to 1, and
EPS evaluates to 0.001.
/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:69: Failure
The difference between v_advanced.pose.orientation.x and 0.0 is nan, which exceeds EPS, where
v_advanced.pose.orientation.x evaluates to -nan,
0.0 evaluates to 0, and
EPS evaluates to 0.001.
/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:70: Failure
The difference between v_advanced.pose.orientation.y and 0.0 is nan, which exceeds EPS, where
v_advanced.pose.orientation.y evaluates to -nan,
0.0 evaluates to 0, and
EPS evaluates to 0.001.
/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:71: Failure
The difference between v_advanced.pose.orientation.z and 0.0 is nan, which exceeds EPS, where
v_advanced.pose.orientation.z evaluates to -nan,
0.0 evaluates to 0, and
EPS evaluates to 0.001.
/home/rosbuild/ci_scripts/ws/src/ros2/geometry2/tf2_geometry_msgs/test/test_tf2_geometry_msgs.cpp:72: Failure
The difference between v_advanced.pose.orientation.w and 1.0 is nan, which exceeds EPS, where
v_advanced.pose.orientation.w evaluates to -nan,
1.0 evaluates to 1, and
EPS evaluates to 0.001.
[  FAILED  ] TfGeometry.Frame (2 ms)
[ RUN      ] TfGeometry.Vector
[       OK ] TfGeometry.Vector (0 ms)
[ RUN      ] TfGeometry.Point
[       OK ] TfGeometry.Point (0 ms)
[----------] 3 tests from TfGeometry (2 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 1 test case ran. (2 ms total)
[  PASSED  ] 2 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TfGeometry.Frame

 1 FAILED TEST
-- run_test.py: return code 1
-- run_test.py: verify result file '/home/rosbuild/ci_scripts/ws/build/tf2_geometry_msgs/test_results/tf2_geometry_msgs/test_tf2_geometry_msgs.gtest.xml'
```

-- http://ci.ros2.org/view/nightly/job/nightly_linux-aarch64_release/76/testReport/(root)/projectroot/test_tf2_geometry_msgs/

wjwwood, 2017-06-19:
This also affects Windows debug nightly.

wjwwood, 2017-06-19:
I believe @tfoote is having a look into it.

dhood, 2017-06-19:
I suspect this is similar to https://github.com/ros2/geometry2/pull/18

mikaelarguedas, 2017-06-20:
@wjwwood I think this can be closed now that the missing initialization has been addressed

wjwwood, 2017-06-20:
Yup, thanks.

mikaelarguedas, 2017-06-20:
acually looks like the test still fails on windows Debug: http://ci.ros2.org/view/nightly/job/nightly_win_deb/513/testReport/. Doesnt seem to be the same issue (looks like a sign issue not uninitialized memory) so I let you decide how you want to track it

wjwwood, 2017-06-20:
Ah you're right. I'm going to reopen this one, since it is the same tests.

@tfoote @dhood if you've seen this one too, let me know.

tfoote, 2017-06-20:
I haven't seen that one. It looks to have regressed in build 507 from the Jun 14, 2017 7:30:00 AM http://ci.ros2.org/view/nightly/job/nightly_win_deb/507/#showFailuresLink

http://ci.ros2.org/view/nightly/job/nightly_win_deb/507/testReport/junit/(root)/TfGeometry/Frame/
```
C:\J\workspace\nightly_win_deb\ws\src\ros2\geometry2\tf2_geometry_msgs\test\test_tf2_geometry_msgs.cpp:54
The difference between v_simple.pose.position.x and -9 is 17.999999999999996, which exceeds EPS, where
v_simple.pose.position.x evaluates to 8.9999999999999964,
-9 evaluates to -9, and
EPS evaluates to 0.001.
```

It does appear that all the values are coming out with the negative of the expected sign for those specific tests.

PR #28 (2017-06-23): initialize whole quaternion
Fixes #24 

This was only showing on Windows Debug builds. But was the same problem as #18 and #25 
It was very odd that it gave exactly 180 responses to what were expected. But also that the other arches passed is surprising too.

CI: 

- Linux: [image]
- linux aarch64 [image]
- osx [image]
- windows [image]
closed 2017-06-23 by wjwwood