ros2/geometry2 #31: `tf_monitor` not seeing transformations published by `static_transform_publisher`

Kukanani, 2017-06-27:
[Windows, FastRTPS]
When running `static_transform_publisher`, `tf_monitor` does not list any provided transforms:

```
> static_transform_publisher 1 2 3 0.1 0.1 0.1 foo bar

LOOPING due to no latching at the moment
LOOPING due to no latching at the moment
...
```

```
> tf2_monitor

RESULTS: for all Frames

Frames:

All Broadcasters:

RESULTS: for all Frames

Frames:

All Broadcasters:

...
```

clalancette, 2017-06-27:
I'm also seeing this on Linux and FastRTPS on arm64.  You beat me to ticketing it :).

Kukanani, 2017-06-27:
`tf2_monitor` is not processing callbacks for the `tf` and `tf_static` topics, despite messages being published there. The node runs two threads, one for spinning and one for sleeping and printing the state of the TF tree. This is because each thread creates its own NodeHandle, and the `spin` call only spins one of them. I have a fix in the works.

closed 2017-06-28 by Kukanani