ros2/geometry2 #34: time.h durationToSec seems very inefficient

nate-jackson, 2017-07-13:
We have been investigating why we can't seem to get lookupTransform to work properly for us, and came across this function.

Instead of doing the big long thing that is going on there:

```
inline double durationToSec(const tf2::Duration& input){
    int64_t count = input.count();
    int32_t sec, nsec;
    nsec = static_cast<int32_t>(count % 1000000000l);
    sec = static_cast<int32_t>((count - nsec) / 1000000000l);

    double sec_double, nsec_double;
    nsec_double = 1e-9 * static_cast<double>(nsec);
    sec_double = static_cast<double>(sec);
    return sec_double + nsec_double;
  }
```

couldn't we just do

```
inline double durationToSec(const tf2::Duration& input){
    return input.count() * 1e-9;
  }
```

I can't see a reason for the extra stuff that is going on in there.

wjwwood, 2017-07-13:
Sounds reasonable to me, it's probably left over from the conversion. @dhood what do you think?

dhood, 2017-07-14:
the proposed conversion isn't as accurate. We might still prefer it (since the conversion isn't exact as is), but in that case we'd need to relax the tolerance on the tests (e.g. [this one](https://github.com/ros2/geometry2/blob/a275730813b9cf55943f3f05d57aaee51b32944a/tf2/test/simple_tf2_core.cpp#L202)) and disable [this test](https://github.com/ros2/geometry2/blob/a275730813b9cf55943f3f05d57aaee51b32944a/tf2/test/simple_tf2_core.cpp#L177)

dhood, 2017-07-14:
Here's CI to show what I mean (these tests were run with `--retest-until-fail 20` because the reduced accuracy would manifest as a flaky test since it wouldn't fail for all values of 'now'):

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

wjwwood, 2017-07-14:
Ah, @dhood I forgot about that. Perhaps we should put a comment explaining why the nanoseconds are scaled separately.

dhood, 2017-07-14:
sure thing: https://github.com/ros2/geometry2/pull/35

dhood, 2017-07-14:
I'll close this because I believe the question has been addressed. if that's not the case feel free to re-open

closed 2017-07-14 by dhood