ros2/geometry2 #38: lookupTransform outputs a transform with wrong timestamp

vsherrod, 2017-08-09:
lookupTransform outputs a transform whose timestamp does not match the time that has been asked for.  

We see the transform's timestamp looping from being 0.5s behind the requested timestamp to being up to 4s ahead of it.  These numbers our of course with our specific setup.  We have the tf being broadcast at 50 hz and it being queried at 4 hz.  

The broadcasting and listening code we are using can be found in the following files:

[tf_listener.txt](https://github.com/ros2/geometry2/files/1212917/tf_listener.txt)

[tf_broadcaster.txt](https://github.com/ros2/geometry2/files/1212930/tf_broadcaster.txt)

Here is the sample output of the listener:

[tf_listener_output.txt](https://github.com/ros2/geometry2/files/1212933/tf_listener_output.txt)

bponsler, 2017-08-09:
@vsherrod -- I hadn't noticed this issue while using rviz in ROS 2, or any other nodes/packages for that matter. I definitely see the same results when running your examples on my setup (thanks for the example code).

I added a print to the `tf2/buffer_core.cpp` to print out when it receives a new transform (the `setTransformImpl` function) and it appears to be receiving the transforms at a fast rate, example:

    BufferCore: GOT TF : 1502319493.572400331
    BufferCore: GOT TF : 1502319493.592642546
    BufferCore: GOT TF : 1502319493.612868309
    BufferCore: GOT TF : 1502319493.633083344
    BufferCore: GOT TF : 1502319493.653306484
    BufferCore: GOT TF : 1502319493.673531532
    BufferCore: GOT TF : 1502319493.693761826
    BufferCore: GOT TF : 1502319493.713982582
    BufferCore: GOT TF : 1502319493.734198093
    BufferCore: GOT TF : 1502319493.754431725
    BufferCore: GOT TF : 1502319493.774652481
    BufferCore: GOT TF : 1502319493.794873476
    BufferCore: GOT TF : 1502319493.815122366
    TimeRequested: 1502319493.795506239
    TimeReceived : 1502319494.564988613
    Diff_Requested_Received: -0.769482374
    
    BufferCore: GOT TF : 1502319493.835351467
    BufferCore: GOT TF : 1502319493.855565310
    BufferCore: GOT TF : 1502319493.875795364
    BufferCore: GOT TF : 1502319493.896017551
    BufferCore: GOT TF : 1502319493.916233540
    BufferCore: GOT TF : 1502319493.936518431
    BufferCore: GOT TF : 1502319493.956826210
    BufferCore: GOT TF : 1502319493.977041721
    BufferCore: GOT TF : 1502319493.997262001
    BufferCore: GOT TF : 1502319494.017485857
    BufferCore: GOT TF : 1502319494.037718534
    BufferCore: GOT TF : 1502319494.058018208
    TimeRequested: 1502319494.045734882
    TimeReceived : 1502319495.815217257
    Diff_Requested_Received: -1.769482374

Note that the GOT TF prints are filtered specifically for `map -> base_link` (which is what tf_listener is looking for).

I would suspect that if an old transform (cycling from 4s to -0.5s) was being returned that the cyclical behavior would be visible in rviz while driving around a robot, but I could wrong. Maybe `lookupTransform()` is returning the latest transform, but somehow returning the incorrect time?

allenh1, 2017-08-09:
```c++
 transform = tf_buffer.lookupTransform("base_link", "map", time, std::chrono::nanoseconds(3000000000));
```

Isn't that a 3 second transform timeout?

bponsler, 2017-08-09:
@allenh1 Yes, it's a three second timeout which is documented as:

    \param timeout How long to block before failing

I think the three seconds timeout will only happen if a transform is not available -- if there's an available transform it should return immediately.

I just tested changing the timeout to 0.1 seconds, and the same cycling still occurs.

    TimeRequested: 1502322524.547150135
    TimeReceived : 1502322524.069718599
    Diff_Requested_Received: 0.477431536
    TimeRequested: 1502322524.797379494
    TimeReceived : 1502322524.319947720
    Diff_Requested_Received: 0.477431774
    TimeRequested: 1502322525.047592402
    TimeReceived : 1502322525.570160866
    Diff_Requested_Received: -0.522568464
    TimeRequested: 1502322525.297808170
    TimeReceived : 1502322525.820376635
    Diff_Requested_Received: -0.522568464
    TimeRequested: 1502322525.547969103
    TimeReceived : 1502322526.070537329
    Diff_Requested_Received: -0.522568226
    TimeRequested: 1502322525.798191309
    TimeReceived : 1502322526.320759773
    Diff_Requested_Received: -0.522568464
    TimeRequested: 1502322526.048437595
    TimeReceived : 1502322527.571005821
    Diff_Requested_Received: -1.522568226
    TimeRequested: 1502322526.298645735
    TimeReceived : 1502322527.821213961
    Diff_Requested_Received: -1.522568226
    TimeRequested: 1502322526.548882008
    TimeReceived : 1502322528.071450472
    Diff_Requested_Received: -1.522568464
    TimeRequested: 1502322526.799113512
    TimeReceived : 1502322528.321681976
    Diff_Requested_Received: -1.522568464
    TimeRequested: 1502322527.049338579
    TimeReceived : 1502322529.571906805
    Diff_Requested_Received: -2.522568226
    TimeRequested: 1502322527.299566746
    TimeReceived : 1502322529.822135210
    Diff_Requested_Received: -2.522568464
    TimeRequested: 1502322527.549809456
    TimeReceived : 1502322530.072377682
    Diff_Requested_Received: -2.522568226
    TimeRequested: 1502322527.800004482
    TimeReceived : 1502322530.322572708
    Diff_Requested_Received: -2.522568226
    TimeRequested: 1502322528.050235510
    TimeReceived : 1502322531.572803736
    Diff_Requested_Received: -3.522568226
    TimeRequested: 1502322528.300439358
    TimeReceived : 1502322531.823007584
    Diff_Requested_Received: -3.522568226
    TimeRequested: 1502322528.550649881
    TimeReceived : 1502322532.073218346
    Diff_Requested_Received: -3.522568464
    TimeRequested: 1502322528.800866604
    TimeReceived : 1502322528.028467655
    Diff_Requested_Received: 0.772398949
    TimeRequested: 1502322529.051067114
    TimeReceived : 1502322529.278668165
    Diff_Requested_Received: -0.227601051

vsherrod, 2017-08-10:
@allenh1 and @bponsler. Yes, that is a 3 second timeout.  I have also run it with varying timeouts as well with similar results as @bponsler has verified.  In fact, on the actual robot, we do not have a timeout since it is asking for a transform of the timestamp of sensor data that is already enough in the past that it should exist in the buffer.  We do not see anything too strange in rviz with the TFs specifically as the robot moves, but I think it is because the latest TF is keeping up with the current time.  The issue is that we want to correctly place obstacles viewed by a sensor into the map frame.  It is very important that we have the correct transform at the time the sensor data was received so it is placed correctly in the map.  The fact that we are not getting a transform with the requested timestamp is troubling. 

@bponsler Are the printouts that you added of the timestamp of the transform it received, or just the time it received the transform in the setTransformImpl function (from something like std::chrono::system_clock::now())?   It seems odd to me that the timestamp of the transform received is ahead of the time of the latest of these printouts.  How is it grabbing a transform whose timestamp is ahead of the latest one received in the buffer?   It would be actually pretty close to the requested time if it had the timestamp of the latest printout.

@bponsler theory of it just grabbing the latest transform could be correct.  I ran a test where I queried for the latest transform as well (by setting the time to 0 as mentioned in the documentation).  The latest transform is always slightly ahead of the received transform if I query for it after.  It is always slightly behind it if I do the query for the latest transform before.  

Results from querying latest transform before the time requested transform:

TimeRequested: 1502379102.020399094
TimeReceived : 1502379106.233688116
LatestTFTime : 1502379106.230337143
Diff_Requested_Received: -4.213289022
Diff_Latest_Received: -0.003350973

TimeRequested: 1502379102.270644903
TimeReceived : 1502379102.188966751
LatestTFTime : 1502379102.178290367
Diff_Requested_Received: 0.081678152
Diff_Latest_Received: -0.010676384

TimeRequested: 1502379102.520904541
TimeReceived : 1502379102.439226151
LatestTFTime : 1502379102.420646429
Diff_Requested_Received: 0.081678391
Diff_Latest_Received: -0.018579721

TimeRequested: 1502379102.771198988
TimeReceived : 1502379102.689520597
LatestTFTime : 1502379102.683475256
Diff_Requested_Received: 0.081678391
Diff_Latest_Received: -0.006045341

TimeRequested: 1502379103.021399736
TimeReceived : 1502379103.939721584
LatestTFTime : 1502379103.925678015
Diff_Requested_Received: -0.918321848
Diff_Latest_Received: -0.014043570

TimeRequested: 1502379103.271556139
TimeReceived : 1502379104.189877748
LatestTFTime : 1502379104.188467026
Diff_Requested_Received: -0.918321609
Diff_Latest_Received: -0.001410723

TimeRequested: 1502379103.521806002
TimeReceived : 1502379104.440127850
LatestTFTime : 1502379104.430803537
Diff_Requested_Received: -0.918321848
Diff_Latest_Received: -0.009324312

TimeRequested: 1502379103.772020817
TimeReceived : 1502379104.690342426
LatestTFTime : 1502379104.673150301
Diff_Requested_Received: -0.918321609
Diff_Latest_Received: -0.017192125

TimeRequested: 1502379104.022263288
TimeReceived : 1502379105.940584898
LatestTFTime : 1502379105.935645342
Diff_Requested_Received: -1.918321609
Diff_Latest_Received: -0.004939556

TimeRequested: 1502379104.272528887
TimeReceived : 1502379106.190850496
LatestTFTime : 1502379106.177727699
Diff_Requested_Received: -1.918321609
Diff_Latest_Received: -0.013122797

Results from querying latest transform after the time requested transform:

TimeRequested: 1502379217.736503363
TimeReceived : 1502379220.985675097
LatestTFTime : 1502379221.005356789
Diff_Requested_Received: -3.249171734
Diff_Latest_Received: 0.019681692

TimeRequested: 1502379217.986671448
TimeReceived : 1502379221.235843182
LatestTFTime : 1502379221.247523785
Diff_Requested_Received: -3.249171734
Diff_Latest_Received: 0.011680603

TimeRequested: 1502379218.236891747
TimeReceived : 1502379218.191096067
LatestTFTime : 1502379218.195081949
Diff_Requested_Received: 0.045795679
Diff_Latest_Received: 0.003985882

TimeRequested: 1502379218.487140417
TimeReceived : 1502379218.441344738
LatestTFTime : 1502379218.458114624
Diff_Requested_Received: 0.045795679
Diff_Latest_Received: 0.016769886

TimeRequested: 1502379218.737431765
TimeReceived : 1502379218.691636562
LatestTFTime : 1502379218.700951815
Diff_Requested_Received: 0.045795202
Diff_Latest_Received: 0.009315252

TimeRequested: 1502379218.987675667
TimeReceived : 1502379218.941879988
LatestTFTime : 1502379218.943386555
Diff_Requested_Received: 0.045795679
Diff_Latest_Received: 0.001506567

TimeRequested: 1502379219.237847090
TimeReceived : 1502379220.192051411
LatestTFTime : 1502379220.206526041
Diff_Requested_Received: -0.954204321
Diff_Latest_Received: 0.014474630

We had this same test working in ROS1, so we will start looking at differences in the code between the two and see if we can spot any significant reason why we are seeing what we are seeing.

bponsler, 2017-08-10:
@vsherrod Here's the code I added to the `BufferCore::setTransformImpl` function in `tf2/src/buffer_core.cpp`

    if (frame_id.compare("map") == 0 && child_frame_id.compare("base_link") == 0) {
      printf("BufferCore: GOT TF : %.9f\n", tf2::timeToSec(stamp));
    }

`stamp` is input to the function from `setTransform` and is the TimePoint equivalent of the transform header stamp.

**TL;DR: I think the `lookupTransform` conversion from TimePoint to message stamp is not correct**

Notes/observations from looking into `lookupTransform` a bit:

- lookupTransform first calls `canTransform`, which appears to typically have one or more extrapolation errors before succeeding (after it receives a newer transform than the requested time)
- canTransform uses `walkToTopParent`
- lookupTransform then calls `lookupTransformImpl` (which also calls walkToTopParent)
- walkToTopParent appears to be returning the time that is passed to it (it uses the TransformAccum object, and sets its time through calls to the `finalize` function which it always passes `time` to)
- lookupTransform receives the correct time in time_out

As far as I can tell, this means that the way `lookupTransform` is converting from time_out to msg.header.stamp is not working correctly.

The current code to convert is:

    std::chrono::time_point_cast<std::chrono::seconds>(time_out);                                                       
    msg.header.stamp.sec = (uint32_t)std::chrono::time_point_cast<std::chrono::seconds>(time_out).time_since_epoch().count();                                                                                                                
    msg.header.stamp.nanosec = (uint32_t)std::chrono::time_point_cast<std::chrono::nanoseconds>(time_out).time_since_epoch().count() - msg.header.stamp.sec;

As a test, I switched to using the code that is used in `tf2_ros::toMsg` found in `tf2_ros/buffer_interface.h`. This code is:

    std::chrono::nanoseconds ns = std::chrono::duration_cast<std::chrono::nanoseconds>(time_out.time_since_epoch());
    std::chrono::seconds s = std::chrono::duration_cast<std::chrono::seconds>(time_out.time_since_epoch());
    msg.header.stamp.sec = (int32_t)s.count();
    msg.header.stamp.nanosec = (uint32_t)(ns.count() % 1000000000ull);

This appears to have fixed the issue, as shown by the following output:

    TimeRequested: 1502390885.353914499
    TimeReceived : 1502390885.353914499
    Diff_Requested_Received: 0.000000000
    TimeRequested: 1502390885.604130507
    TimeReceived : 1502390885.604130507
    Diff_Requested_Received: 0.000000000
    TimeRequested: 1502390885.854372740
    TimeReceived : 1502390885.854372740
    Diff_Requested_Received: 0.000000000
    TimeRequested: 1502390886.104542017
    TimeReceived : 1502390886.104542017
    Diff_Requested_Received: 0.000000000
    TimeRequested: 1502390886.354771376
    TimeReceived : 1502390886.354771376
    Diff_Requested_Received: 0.000000000

Of course, this assumes that `tf2_ros::toMsg(time)` is doing the conversion correctly :)

vsherrod, 2017-08-10:
Nice! Do you want to do a pull request on this, or do you want us to?

SecretaryBirds, 2017-08-10:
I've been pouring over the TF code, and I don't see any way this bug could have affected retrieving the data you actually cared about. 
If the TimePoint logic is all sound in the TransformStorage, then when you request data, it should retrieve the correct stuff, but when it converts it to a TransformStamped message, the timestamp just gets messed up. 

Is that really the only impact on this? 
@bponsler @allenh1 @tfoote can you see any other impact from this bug?

tfoote, 2017-08-10:
That looks like the right fix, and it does look like it was just the output timestamp conversion, which should not effect the correctness of the other data. It could have effected data if the result was used again as in input for another query or the output timestamp was reused in any way.

tfoote, 2017-08-23:
fixed in #39

closed 2017-08-23 by tfoote