ros2/geometry2 #41: unable to use TransformListener constructor that takes a node

BrannonKing, 2017-08-15:
`TransformListener` has a constructor that takes a Node pointer, but if you use it you get this error: 
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Node has already been added to an executor.
```
Does the listener require a dedicated Node (like the one created in the constructor that doesn't take a node)?

dhood, 2017-08-15:
I have been able to use a listener in a non-dedicated node (in [amcl](https://github.com/ros2/navigation/blob/a9a3f49f4608ba793485f8416adbdbf14f8090a6/amcl/src/amcl_node.cpp)). Your issue might be that by default the listener will call spin underneath. If you're not already, you can disable it by setting `spin_thread` to be false e.g. `tf2_ros::TransformListener(*tf2_buffer_, node, false);`

BrannonKing, 2017-08-15:
That works. Thanks.

closed 2017-08-15 by BrannonKing