ros2/geometry2 #45: Inconsistent usage string among executables

mikaelarguedas, 2017-09-11:
Not sure what to do here.

I started looking at this because the usage string of `tf2_monitor` is missing a trailing newline.
Right now it's using `ROS_INFO` aliased with `printf`: https://github.com/ros2/geometry2/blob/d8958f688bee423dcaa3aa73f4da7a4be0e811d1/tf2_ros/src/tf2_monitor.cpp#L46

We could do one of the following:

- Change to call printf explicitely with a trailing new line like it's done in tf2_echo: https://github.com/ros2/geometry2/blob/d8958f688bee423dcaa3aa73f4da7a4be0e811d1/tf2_ros/src/tf2_echo.cpp#L69

- Switch to ros2 logging but it will make the usage string hard to read because of the current lack of configuration:
```
[INFO] []: TF_Monitor: usage: tf2_monitor framea frameb (main() at /home/mikael/work/ros2/test_sprint_ws/src/ros2/geometry2/tf2_ros/src/tf2_monitor.cpp:290)
```
- Do nothing for now

tfoote, 2017-09-11:
Upstream it's generally using ROS_INFO we talked offline with @wjwwood and @mikaelarguedas about having INFO suppress the filename by default but error or higher show the file and line number since it's less common and won't get in the way as much on the console.

In the long run getting the ROS_INFO to go out via RCUTILS whether via a #define or otherwise to not diverge from upstream too much.

CursedRock17, 2024-03-25:
It's been a while and [that line](https://github.com/ros2/geometry2/blob/rolling/tf2_ros/src/tf2_monitor.cpp#L280) which used `ROS_INFO` now uses `RCLCPP_INFO` is that good enough with a 
> Switch to ros2 logging 

or should it become `RCUTILS`

clalancette, 2024-03-25:
No, `RCLCPP_` are preferred where we can use it, as it gives the full range of logging facilities.

But you are right, this looks like it is done, so I'll close this out.

closed 2024-03-25 by clalancette