ros2/geometry2 #59: Porting tf2_ros/include/message_filter.h to ROS2.

sagniknitr, 2018-06-21:
## Feature request
Can the message_filter.h of tf2_ros be ported to ROS2.0 (https://github.com/ros2/geometry2/blob/ros2/tf2_ros/include/tf2_ros/message_filter.h)? There is already a message_filter package available for ROS2.0 (https://github.com/intel/ros2_message_filters).

nburek, 2018-08-15:
I am actively working on porting message_filter.h to ROS2. It looks like the intel/ros2_message_filters repository just got merged into https://github.com/ros2/message_filters. I'm currently blocked on the significant change in callback interfaces between ROS1 and ROS2 (see: https://discourse.ros.org/t/ros2-package-migration-replacement-for-callbackqueueinterface/5751).

dirk-thomas, 2018-08-27:
> There is already a message_filter package available for ROS2.0 (https://github.com/intel/ros2_message_filters).

The repository containing the `message_filters` package for ROS 2 is https://github.com/ros2/message_filters

Please consider to create a ticket or even better a pull request there.

sagniknitr, 2018-08-28:
@dirk-thomas Thanks for the message filters link. However, the feature request I created is for the file inside tf2_ros package
(https://github.com/ros2/geometry2/blob/ros2/tf2_ros/include/tf2_ros/message_filter.h). Is there a plan to remove the particular file from tf2_ros ?

dirk-thomas, 2018-08-28:
> Can the message_filter.h of tf2_ros be ported to ROS 2.

I see now that the referenced file in this repo wasn't ported. I am not sure what it used exactly but please feel free to provide a pull request to port it to ROS 2 using the now available message_filters repo.

yechun1, 2018-10-18:
If no one started work on it, I would like to port the tf2_ros message_filter.
To both fix navigation dependence issue: https://github.com/ros-planning/navigation2/issues/171

mkhansen-intel, 2018-10-23:
@yechun1 - do you have any update on this?

yechun1, 2018-10-24:
@mkhansen-intel started porting from yesterday, there are several porting issues I'm working on today. Which include connection signals2 issue as https://github.com/ros2/message_filters/issues/11,  ros::Time port to tf2::TimePoint, boost::mutex to std::mutex, etc.

yechun1, 2018-10-24:
Till now, most code ported except `ros::CallbackQueueInterface` related APIs, synce there is no easy way to replace it with ROS2 APIs(not found yet), I need to do more research and change the code. Appreciate if any one could share the idea or reference code to replace `ros::CallbackQueueInterface`

yechun1, 2018-10-26:
Block on https://github.com/ros2/message_filters/issues/11

gaoethan, 2018-10-30:
I had a simple check that there is only the following 2 APIs which uses the callbackqueue of node handle if it's provided. as follows:

```
  void messageDropped(const MEvent& evt, FilterFailureReason reason)
  {
    if (callback_queue_)
    {
      ros::CallbackInterfacePtr cb(new CBQueueCallback(this, evt, false, reason));
      callback_queue_->addCallback(cb, (uint64_t)this);
    }
    else
    {
      signalFailure(evt, reason);
    }
  }

  void messageReady(const MEvent& evt)
  {
    if (callback_queue_)
    {
      ros::CallbackInterfacePtr cb(new CBQueueCallback(this, evt, true, filter_failure_reasons::Unknown));
      callback_queue_->addCallback(cb, (uint64_t)this);
    }
    else
    {
      this->signalMessage(evt);
    }
  }
```

So for 2 potential workaround to fix the lacking support from rclcpp node implementation to callback queue:
1. try to directly replace the callback_queue_ logic here with the current ros2 callback group. for example, direct bind or something similar
2. use the `else` branch by employing  `signalMessage` which will also call the the corresponding callbacks, and that mechanism already works in ros2 message_filters

gaoethan, 2018-11-08:
I've already worked out an available one by: 
 - use rclcpp Time and Duration
 - remove boost with C++11 instead
 - remove the way to using callback queue (it can be a follow-up feature while rclcpp support the similar way of ros callback queue later)
 - extend the simple filter to support the case of failure callback registration

please have a try with the following code base and have a review/fix issue when found:
https://github.com/gaoethan/message_filters.git with branch `master`
https://github.com/gaoethan/geometry2.git with branch `tf2_ros_message_filters`

yechun1, 2018-11-09:
@gaothan, seems not all boost has been removed from the commit, would you please help to check. And suggest to sync latest code from ros2.repo as some core APIs changed to avoid APIs changing impact.

gaoethan, 2018-11-23:
I think that It's already done for this issue, I have clean-up my porting and upstream it  as (https://github.com/ros2/geometry2/pull/81) which is already in geometry2 now. Hopefully this benefits you.

sagniknitr, 2018-11-23:
@gaoethan Can you check this patch (https://github.com/sagniknitr/geometry2/commit/2fd5ab430ec1629ab618755a467a8bbb17d2a559).

There seems to be a error with rclcpp::Clock::now() function call ;  which you are using in the code you pushed. Apparently you need to create a Clock object for using rclcpp::Clock::now().

gaoethan, 2018-11-26:
@sagniknitr Thanks, so... you prefer a dedicated clock object instead of invoking directly. :)

sagniknitr, 2018-11-26:
AFAIK we cannot invoke `now()` directly without creating a clock object. Or has it been changed recently in the rclcpp layer ?

yechun1, 2018-12-26:
This issue should be fixed by https://github.com/ros2/geometry2/pull/81 now.

closed 2018-12-26 by sagniknitr