ros2/geometry2 #68: reuse logic from Time::durationToSec method

dirk-thomas, 2018-08-21:
As suggested in https://github.com/ros2/rclcpp/pull/536#discussion_r211035479 the [logic](https://github.com/ros2/geometry2/blob/7dafb4d6c0c7025dfeeed3cef336113c62601dac/tf2/include/tf2/time.h#L74..L84) could be moved to `rcutils` and be used in `rclcpp::Time ` and `rclcpp::Duration`.

CursedRock17, 2024-03-29:
I can open up a PR for this since it's real simple logic movement but I would I assume it would make more sense to move the logic to [rclcpp](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/duration.cpp#L263) opposed to [rcutils] (https://github.com/ros2/rcutils/blob/rolling/include/rcutils/time.h#L39) to keep rcutils concise. Does anybody else have any opinions on where it should be moved to?

clalancette, 2024-04-03:
> I can open up a PR for this since it's a simple logic movement but I would assume it would make more sense to move the logic to [rclcpp](https://github.com/ros2/rclcpp/blob/rolling/rclcpp/src/rclcpp/duration.cpp#L263) opposed to [rcutils](https://github.com/ros2/rcutils/blob/rolling/include/rcutils/time.h#L39) to keep rcutils concise.

I don't think we can do that.  The problem is that `tf2` does not, and cannot, depend on `rclcpp`.  However, both `tf2` and `rclcpp` currently depend on `rcutils`, so that is a logical place to share the implementation.