ros2/geometry2 #78: Wrong package.xml file in tf2_sensor_msgs

paoloach, 2018-11-02:
The CMakeLists.txt require the Eigen3 library while in the package.xml file is declared as eigen

## Bug report

**Required Info:**

- Operating System:
  - gentoo with profile amd64/17.0/desktop/plasma
- Installation type:
  - from source  
- Version or commit hash:
  - db72716cf2cf2ae55561fc1db01ee2c63ea6414b
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  -  N/A 

#### Steps to reproduce issue
following the instruction for install ROS2 as 
  mkdir -p ~/ros2_ws/src
  cd ~/ros2_ws
  wget https://raw.githubusercontent.com/ros2/ros2/master/ros2.repos
  vcs import ~/ros2_ws/src < ros2.repos
  colcon build --symlink-install

#### Expected behavior
tf2_sensor_msgs build and install

#### Actual behavior
tf2_sensor_msgs fails configuration phase because Eigen3 is not declared

mjcarroll, 2018-11-02:
`eigen` is the correct rosdep key in this case: https://github.com/ros/rosdistro/blob/0d0b72dc8853d6aa38e739f761cd7ef2edec30ab/rosdep/base.yaml#L649

Have you correctly installed the rosdeps?

paoloach, 2018-11-03:
Hi mjcarroll,

honestly, I'm new on ROS and and so I'm not sure that rosdeps is correctly installed (the colcon tutorial how to install ROS2 doesn't mention it)

But, what it confuses me is that
1) in the CMakeFile there is a dependency to Eigen3 while in the package.xml the dependency is to eigen
2) checking my system, I have eigen but I don't found Eigen3 but the package compile only depending on Eigen3.

Anyway, you know better that me the package so if you state that it is right have Eigen3 in the CMakeFile and eigen in the package.xml file, well, it should be my gentoo distrubution that has some problems.

Regards,
Paolo

jacobperron, 2018-11-05:
@paoloach You are missing the rosdep commands for installing dependencies between the `vcs import` and `colcon build` commands:

From the [documentation](https://index.ros.org/doc/ros2/Linux-Development-Setup/#install-dependencies-using-rosdep):

    sudo rosdep init
    rosdep update
    rosdep install --from-paths src --ignore-src --rosdistro bouncy -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 rti-connext-dds-5.3.1 urdfdom_headers"

The `rosdep init` command only needs to be run once, but the other two commands should be run after  updating source where changes to package dependencies (e.g. in package.xml) occurs.

mjcarroll, 2018-11-08:
I'm closing this, because I don't believe this is a bug, @paoloach feel free to reopen if you discover a bug.

closed 2018-11-08 by mjcarroll