ros2/geometry2 #80: static_transform_publisher can't handle __params:= or __name:= given from launch file

mkhansen-intel, 2018-11-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - ros2 branch
- DDS implementation:
  - default / FastRTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
The argument parsing in static_transform_publisher can't currently handle the params passed by the launch file:

```
ros2 launch nav2_bringup test_bringup_launch.py
```
test_bringup_launch.py file is this:
```
from launch import LaunchDescription

import launch.actions
import launch_ros.actions

def generate_launch_description():
    params_file = launch.substitutions.LaunchConfiguration('params', default='nav2_params.yaml')
    
    return LaunchDescription([

        launch.actions.DeclareLaunchArgument(
            'params', default_value='nav2_params.yaml', description='Full path to params file to load'),
        
        launch_ros.actions.Node(
            package='tf2_ros',
            node_executable='static_transform_publisher',
            output='screen',
            arguments=['0', '0', '0', '0', '0', '0', 'base_link', 'base_footprint'],
            parameters=[params_file])
    ])

```
Params file is this:
```
static_transform_publisher:
    ros__parameters:
        use_sim_time: True
```
#### Expected behavior
The parameter is set correctly and the TF is published correctly.
#### Actual behavior
The parameter is set correctly.

However the TF is not published correctly. When I view the TF using this command:
```
ros2 run tf2_ros tf2_echo base_link base_footprint
```
I get these errors:
```
Failure at 1.54169e+09
Exception thrown:"base_link" passed to lookupTransform argument target_frame does not exist. 
The current list of frames is:

Error:   TF_NAN_INPUT: Ignoring transform for child_frame_id "__params:=nav2_params.yaml" from authority "Authority undetectable" because of a nan value in the transform (0.000000 0.000000 0.000000) (-nan -nan -nan -nan)
         at line 258 in /home/mkhansen/ros2_dev/ros2_ws/src/ros2/geometry2/tf2/src/buffer_core.cpp
Error:   TF_DENORMALIZED_QUATERNION: Ignoring transform for child_frame_id "__params:=nav2_params.yaml" from authority "Authority undetectable" because of an invalid quaternion in the transform (-nan -nan -nan -nan)
         at line 271 in /home/mkhansen/ros2_dev/ros2_ws/src/ros2/geometry2/tf2/src/buffer_core.cpp
```

#### Additional information
I looked at the static_transform_broadcaster_program.cpp and can see where it is parsing the arguments. It is currently looking at 'argc' to decide how to assign the variables. It would need to first strip the __params:= and __name:= from the args list before proceeding.

sloretz, 2018-11-08:
Could be solved by `static_transform_broadcaster_program.cpp` using `init_and_remove_ros_arguments()` or `remove_ros_arguments()`

https://github.com/ros2/rclcpp/blob/33a755c535d654c97401eb6d199580368c19eb40/rclcpp/include/rclcpp/utilities.hpp#L65-L82

mkhansen-intel, 2019-02-09:
Did this get merged and fixed for Crystal? It looks to me like it did but it's still marked open so I wasn't sure.

clalancette, 2019-02-09:
@mkhansen-intel Yeah, it looks like the code for this got merged.  Can you confirm that your original issue is solved?  Assuming so, feel free to close this out.

clalancette, 2019-09-16:
I think this is solved as of Crystal, so I'm going to close it out.  Feel free to open a new issue if you are still having problems.

closed 2019-09-16 by clalancette