ros2/geometry2 #87: tf2_ros is not built for Python

JacobJBeck, 2018-12-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 LTS
- Installation type:
  - Binaries or source
- Version or commit hash:
  - d89b9a2e154a19c224dade0b13275df34180bf12
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclpy

#### Steps to reproduce issue

```
import tf2_ros
```

#### Expected behavior
Module `tf2_ros` is imported.

#### Actual behavior
Error `ModuleNotFoundError: No module named 'tf2_ros'` is raised.

#### Additional information
A pull request for this issue [is here](https://github.com/ros2/geometry2/pull/26).

JacobJBeck, 2018-12-12:
I modified and tried to build the linked fork, but it didn't resolve the issue. My repos file is attached below.

[ros2.repos](https://github.com/ros2/geometry2/files/2673038/ros2.repos.txt)

clalancette, 2018-12-12:
Yeah, this is one part of geometry2 that we haven't ported yet.  PR #26 that you linked to needs some work to get it into shape for merging.  If you'd like to see it in ROS 2 sooner rather than later, consider updating that PR to the newest code and responding to the feedback from review.  For now, I'll mark this as an enhancement.

vinnamkim, 2019-03-15:
@clalancette I'd like to take on this issue. Is it possible?

clalancette, 2019-03-15:
> @clalancette I'd like to take on this issue. Is it possible?

@vinnamkim Feel free to fork off #26 and then make changes.  From there, you can open a PR targeting #26, and we can merge it into that PR to finish the review there.

clalancette, 2019-09-16:
#99 is now merged, so closing this.

closed 2019-09-16 by clalancette