ros2/geometry2 #92: threading error spews a non-stop stream of messages via printf

mkhansen-intel, 2019-02-09:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
- Installation type:
Source
- Version or commit hash:
ros2 branch, 0.10.1 tag
- DDS implementation:
Fast RTPS
- Client library (if applicable):
rclcpp

#### Steps to reproduce issue

```

```
Not sure how exactly to reproduce as I don't fully understand yet. 

#### Expected behavior

#### Actual behavior
Getting this message spewed to the screen continuously:
```
Do not call canTransform or lookupTransform with a timeout unless you are using another thread for populating data. Without a dedicated thread it will always timeout.  If you have a seperate thread servicing tf messages, call setUsingDedicatedThread(true) on your Buffer instance.
```
Warning: Do not taunt [happy fun ball!](https://en.wikipedia.org/wiki/Happy_Fun_Ball) :)

#### Additional information
Looking at the code, I see that first off this is using ROS_ERROR to print, which is #defined as 'printf', with a TODO to fix it. 
https://github.com/ros2/geometry2/blob/d89b9a2e154a19c224dade0b13275df34180bf12/tf2_ros/src/buffer.cpp#L212
``` C++
//TODO(tfoote replace these terrible macros)
#define ROS_ERROR printf
#define ROS_FATAL printf
#define ROS_INFO printf
#define ROS_WARN printf
```

As for the message itself, what does it really mean? I'm not sure what is causing it exactly except that I get it here:
```
  node_handle_ = std::shared_ptr<::rclcpp::Node>(this, [](::rclcpp::Node *) {});

  constexpr double kTfBufferCacheTimeInSeconds = 10.;
  tf_broadcaster_.reset(new tf2_ros::TransformBroadcaster(node_handle_));
  tf_buffer_.reset(new tf2_ros::Buffer(get_clock(), tf2::durationFromSec(kTfBufferCacheTimeInSeconds)));
 tf_listener_.reset(new tf2_ros::TransformListener(*tf_buffer_));
```
When I change the last line to be this to re-use the existing node instead of creating a new one:
```
  tf_listener_.reset(new tf2_ros::TransformListener(*tf_buffer_, node_handle_, false))
```
Then I get this error. 

I have to make this change, or I get an exception thrown by the logger due to duplicate node names. 
https://github.com/ros2/rcl/issues/375

Please let me know how to properly declare a TransformListener so that I don't get this error.

mkhansen-intel, 2019-02-09:
Adding @crdelsey to add any additional info.

tfoote, 2019-02-14:
This is related to #89 that we don't have the ability to have custom callback queues. So the listener is starting it's own node to make sure that things are going to come in while it's blocking the callback thread. You can use the false constructor argument as long as you make sure that there's another executor with a thread that will be servicing incoming tf messages. And in that case you should tell the Buffer that you have another thread available using the `setUsingDedicatedThread(true)` method.

cottsay, 2019-02-21:
@mkhansen-intel, were you able to work around the errors using @tfoote's suggestions?

mkhansen-intel, 2019-03-07:
@cottsay - we were able to work around this in our stack, I'm no longer seeing this error. 

However, it seems like an issue is still needed to fix the fact that printf is being used instead of the RCLCPP_ERROR logging function.

mjcarroll, 2019-03-07:
Thanks @mkhansen-intel, I captured the `printf` defect in #97, so I'll close this one.

closed 2019-03-07 by mjcarroll