ros2/geometry2 #94: LifecycleNode support for tf2_ros::TransformListener

mjeronimo, 2019-03-04:
## Feature request

#### Feature description
Similar to #70, the TransformListener class needs support for lifecyclde nodes. 

#### Implementation considerations
There could be a constructor that takes the required node interfaces, perhaps in addition to rclcpp::Node and rclcpp_lifecycle::LifecycleNode.

mjeronimo, 2019-03-07:
@wjwwood @tfoote I'd like to update the TransformListener class to work with interface pointers rather than rclcpp::Node::SharedPtr so that it can work with lifecycle nodes. However, it's not clear to me how one would use the node_topics interface to create a subscription. It requires a subscription factory and subscription options from the RCL layer. I can't seem to untangle this. Can you provide some guidance on how this should be done? Thanks.

kurshakuz, 2021-09-19:
I believe that issue was fixed in https://github.com/ros2/geometry2/pull/108

DLu, 2022-08-08:
@mjeronimo Can this be closed?

mjeronimo, 2022-08-09:
@DLu Yes, thanks for following up.

closed 2022-08-26 by tfoote