ros2/geometry2 #95: Lifecycle node support for tf2_ros::MessageFilter

mjeronimo, 2019-03-04:
## Feature request

#### Feature description
Similar to #70 and #94, the MessageFilter class needs support for lifecycle nodes. 

#### Implementation considerations
There could be a constructor that takes the required node interfaces, perhaps in addition to rclcpp::Node and rclcpp_lifecycle::LifecycleNode.

kurshakuz, 2021-09-19:
Are there any updates on that?

clalancette, 2021-09-24:
> Are there any updates on that?

Unfortunately not; this is still a backlog/help wanted item.  We'd be happy to review contributions that improve it, though.

CursedRock17, 2024-03-10:
According to [this table](https://github.com/ros-planning/navigation2/issues/608) this issue was solved with issue #96. Is that true or can I pick this up?

closed 2025-08-27 by ahcorde