ros2/geometry2 #110: Finish porting Python portions of tf2_kdl

clalancette, 2019-05-09:
Currently the Python portions of tf2_kdl are commented out: https://github.com/ros2/geometry2/pull/90/files#diff-61737c5a9ef69ef2118eafbc59732b6aR21 .  This is because at the time the tf2_kdl port was done, the tf2_py package was not yet ported (see #99).  Once that is done, we can finish the port of the tf2_kdl python portions.

@mlautman has additional followup tasks (mostly around the python) in https://github.com/ros2/geometry2/pull/90#issuecomment-474060368, which should be completed when we complete the python port.

clalancette, 2021-10-21:
> Can this be ported now?

Unfortunately this is blocked on https://github.com/ros2/orocos_kinematics_dynamics/pull/19

Bardhitoo, 2024-01-26:
@clalancette Are we there yet?

clalancette, 2024-01-26:
> @clalancette Are we there yet?

I think so; we now have PyKDL available.  So now this just needs someone willing to do the work to do the port of the code.