ros2/geometry2 #116: Not finding dashing tf2_kdl on debians

anasarrak, 2019-06-12:
I'm not able to install ``ros-dashing-tf2-kdl`` from debians. 

```
erle@kaladin:/opt/ros/dashing$ sudo apt install ros-dashing-tf2-
ros-dashing-tf2-dbgsym         ros-dashing-tf2-msgs           ros-dashing-tf2-ros-dbgsym
ros-dashing-tf2-eigen          ros-dashing-tf2-msgs-dbgsym    ros-dashing-tf2-sensor-msgs
ros-dashing-tf2-geometry-msgs  ros-dashing-tf2-ros 
```
But installying it from source it compiles and works perfectly.

vmayoral, 2019-06-12:
Related to https://github.com/ros2/geometry2/pull/90. I can confirm I'm experiencing the same issue.

dirk-thomas, 2019-06-12:
The reason why it is not released is that the package is being ignore in the release repo: https://github.com/ros2-gbp/geometry2-release/blob/864cb8cd7947f3b90d2f709890601f699322f4c7/dashing.ignored#L5

@tfoote Can you clarify if that is intentional? If yes, please describe why that is the case? If no, please do a new release with updated ignores.

clalancette, 2019-06-12:
I've done it in https://github.com/ros2-gbp/geometry2-release/pull/2

anasarrak, 2019-06-17:
Thanks @clalancette and @dirk-thomas, this issue has been addresed on the [Patch Release 1](https://github.com/ros2/ros2/releases/tag/release-dashing-20190614). Closing this issue.

closed 2019-06-17 by anasarrak