ros2/geometry2 #125: Future of tf2::convert in ROS2

jdlangs, 2019-07-05:
The ROS2 port has had the heterogeneous `convert` function disabled for quite a while now due to the inability to distinguish ROS messages at compile time (see [here](https://github.com/ros2/geometry2/blob/ca889e48069707330d221392bda8ad99900ce954/tf2/include/tf2/convert.h#L113)). Twice now I've spend a good chunk of time debugging why the call to convert wasn't compiling before finally opening the source and discovering the functionality was removed.

How can this be resolved? If a trait-based conversion system isn't visible on the horizon, should it be declared conversions will have to use `fromMsg`/`toMsg` explicitly?

artivis, 2019-07-12:
Related https://github.com/ros/geometry2/issues/389

sloretz, 2019-10-14:
Related to #176