ros2/geometry2 #128: Support for ROS2 Dashing along with python3

shikharvaish28, 2019-07-16:
```Traceback (most recent call last):
  File "ros_pp.py", line 12, in <module>
    from tf2_ros.src.tf2_ros.transform_listener import TransformListener
  File "../../../src/geometry2/tf2_ros/src/tf2_ros/__init__.py", line 39, in <module>
    from buffer_interface import *
ModuleNotFoundError: No module named 'buffer_interface'
```
I have built the geometry library for my project using `colcon` for ROS2 Dashing and Python 3 and sourced it using `source install/setup.bash` but am getting this error on importing the `TransformListener` from `tf2_ros`. 

My directory structure is as follows:

```
-workspace
         - build
         - install
         - src
                -  my_module
                          - src
                              - file.py
                -  geometry2

```

It would be great if anyone could help me know where exactly the error is coming from.

clalancette, 2019-07-16:
The basic problem is that we haven't ported the python portion of tf2/tf2_ros to ROS 2 yet.  In point of fact, you got further than I think you should have; it should have failed to find the `tf2_ros` module, not just the buffer_interface.  Are you using some custom geometry2 code?

shikharvaish28, 2019-07-16:
Thanks for the clarification @clalancette 

> Are you using some custom geometry2 code?

Yes, I altered a couple of lines thinking that it would work but now it appears that it requires more work.

>  The basic problem is that we haven't ported the python portion of tf2/tf2_ros to ROS 2 yet.

Okay, now I get it. So I guess, I will have to use the tf2 in CPP as of now. Is tf2_ros fully supporting ROS2 dashing?

clalancette, 2019-07-16:
> > 
> 
> Okay, now I get it. So I guess, I will have to use the tf2 in CPP as of now. Is tf2_ros fully supporting ROS2 dashing?

Yes, we've ported the C++ version of tf2_ros to Dashing.  All of the major components are there, but do please let us know if you run into issues (or even better, open a PR to fix them).

I'm going to close this out since it is essentially a duplicate of #99, but feel free to reopen if you need something different.

closed 2019-07-16 by clalancette