ros2/geometry2 #132: Buffer does not appropriately lock BufferCore in `Buffer::lookupTransform`

rotu, 2019-07-22:
In `tf2::Buffer::lookupTransform`, there are consecutive calls to `canTransform` and `lookupTransform`. Internally, both calls take a lock on the underlying `BufferCore` object, but the lack is relinquished in between those two function calls.

This can result in a failure to transform with no error message or an incorrect error message.

tfoote, 2019-07-24:
Please provide a link to the lines in question. I'm assuming you're talking about tf2_ros::Buffer::lookupTransform. 

Can you explain what you consider an "incorrect" error message and what scenario would produce it?

Also in what use case would you get no error message?

rotu, 2019-07-24:
Oops. I was misreading the code here https://github.com/ros2/geometry2/blob/046df60dad728754c434b31f07423fe72db91670/tf2_ros/src/buffer.cpp#L97

`Buffer` has some `canTransform` method, which I thought was being used to generate error messages for why a transform is impossible, but it looks like the call is instead to block until BufferCore has a path between the two frames. `BufferCore` also has some `canTransform` methods which immediately check whether a transform is possible, without blocking.

The `BufferCore` lock is not held between these two calls, and so there's a remote possibility that the TF tree can change in between these methods. But it looks like that shouldn't pose big problems here (though it might in navigation2 where there are some calls to `canTransform` in order to get the error message, with the assumption that it will be the same when the transform is actually attempted)

closed 2019-07-24 by rotu