ros2/geometry2 #141: Raise std::future_error: Promise already satisfied

vinnamkim, 2019-07-31:
## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
- Installation type:
source
- Version or commit hash:
e29db0a46edfd1ccfbf538096ad605359f7a6109
- DDS implementation:
Fast-RTPS
- Client library (if applicable):
rclcpp, nav2_amcl

#### Steps to reproduce issue
Launching nav2_bringup's nav2_simulation_launch.py and waiting some minutes.

#### Expected behavior
Nothing happen

#### Actual behavior
Exception(`what():  std::future_error: Promise already satisfied`) is raised at the following code.
https://github.com/ros2/geometry2/blob/e29db0a46edfd1ccfbf538096ad605359f7a6109/tf2_ros/src/buffer.cpp#L256

#### Additional information
It seems `Buffer::timerCallback` is called
https://github.com/ros2/geometry2/blob/e29db0a46edfd1ccfbf538096ad605359f7a6109/tf2_ros/src/buffer.cpp#L256
after `cb_handle` is called.
https://github.com/ros2/geometry2/blob/e29db0a46edfd1ccfbf538096ad605359f7a6109/tf2_ros/src/buffer.cpp#L205

vinnamkim, 2019-07-31:
Related to https://github.com/ros-planning/navigation2/issues/967

mkhansen-intel, 2019-08-02:
@jacobperron @tfoote - can someone prioritize looking into this, it is causing failures in navigation2 about 50-60% of the time

jacobperron, 2019-08-05:
I also ran into this myself. Working on a fix now.

jacobperron, 2019-08-06:
This bug was introduced in #121.
Fixed in #143.

jacobperron, 2019-08-06:
@vinnamkim @mkhansen-intel Can someone confirm that applying #143 resolves the issue for you?

PR #143 (2019-08-06): Fix race between timeout and transform ready callback
The timer-request handle map is used to check if the promise to the user has already been set.
For the transformable request callback, we loop over the values of the map to check if the request is still valid.
Alternatively, we could implement a bi-directional map for efficiency, but I don't expect the number of entries in the timer-request map to be large so it seems like a pre-mature optimization.

Fixes #141
commit 41460273 (2019-08-06): Fix race between timeout and transform ready callback (#143)

* Add test for race in tf2_ros::Buffer

* Fix race between timeout and transform ready callback

The timer-request handle map is used to check if the promise to the user has already been set.
For the transformable request callback, we loop over the values of the map to check if the request is still valid.
Alternatively, we could implement a bi-directional map for efficiency, but I don't expect the number of entries in
the timer-request map to be large so it seems like a pre-mature optimization.

Fixes #141
closed 2019-08-06 by jacobperron