ros2/geometry2 #142: ros2 branch tf2_ros fails to find BufferCoreInterface on clean build

SteveMacenski, 2019-08-02:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
  - 
- Installation type: source
  - 
- Version or commit hash: ros2 
  - 
- DDS implementation: n/a
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
build from source: colcon build
```

#### Expected behavior
Compiling

#### Actual behavior
Not compiling

#### Additional information

```
/home/s.macenski/Documents/navigation2_ws/src/geometry2/tf2_ros/include/tf2_ros/message_filter.h:131:40: error: ‘BufferCoreInterface’ is not a member of ‘tf2’
     static_assert(std::is_base_of<tf2::BufferCoreInterface, BufferT>::value,

```
----

I've tried to add explicitly the header for buffer core interface (though it should be found) and the error changes to 

```
/home/s.macenski/Documents/navigation2_ws/src/geometry2/tf2_ros/include/tf2_ros/message_filter.h: In instantiation of ‘tf2_ros::MessageFilter<M, BufferT>::MessageFilter(BufferT&, const string&, uint32_t, const SharedPtr&) [with M = geometry_msgs::msg::PointStamped_<std::allocator<void> >; BufferT = tf2_ros::Buffer; std::__cxx11::string = std::__cxx11::basic_string<char>; uint32_t = unsigned int; rclcpp::Node::SharedPtr = std::shared_ptr<rclcpp::Node>]’:
/home/s.macenski/Documents/navigation2_ws/src/geometry2/tf2_ros/test/message_filter_test.cpp:65:90:   required from here
/home/s.macenski/Documents/navigation2_ws/src/geometry2/tf2_ros/include/tf2_ros/message_filter.h:132:5: error: static assertion failed: Buffer type must implement tf2::BufferCoreInterface
     static_assert(std::is_base_of<tf2::BufferCoreInterface, BufferT>::value,
     ^~~~~~~~~~~~~
```

dirk-thomas, 2019-08-02:
The `ros2` branch targets the current `master` of ROS 2 which will become Eloquent in November. If you want to build against Dashing you need to use the `dashing` branch. Or if you really need changes only available on the `ros2` branch you need to build the rest of ROS 2 with the latest state.

SteveMacenski, 2019-08-02:
@dirk-thomas I am building against ros2 master branch 

I am also building the rest of ROS 2 with the latest state, this issue is referring to the fact that tf2_ros is linking against dashing debians and not tf2 in the same workspace

dirk-thomas, 2019-08-02:
Please provide more context and exact steps what you are doing. The information in the original description are not sufficient.

SteveMacenski, 2019-08-05:
I nuked everything and rebuilt, I think I got into a weird state that I cannot now reproduce in an isolated environment. My guess is there was something in a lower workspace I was working with that had a dirty environment.

closed 2019-08-05 by SteveMacenski