ros2/geometry2 #145: Default waitForTransform timeout of 0ns in message filter results in 100% transform timeout

crdelsey, 2019-08-06:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - fccfd98 (latest on master 6 Aug 2019)
- DDS implementation:
  - fastRTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I see this problem when running the navigation2 stack in Gazebo. I can try to provide a simple way to reproduce if needed.

#### Expected behavior
Message filter will wait a reasonable time for the incoming transform

#### Actual behavior
Almost every incoming laser scan message results in the following error message
```
[signalFailure] Drop message: frame 'base_scan' at time 13.201 for reason(0)
```

#### Additional information
nav2_amcl is using message_filters to wait for the transform associated with lidar scan data. With the introduction of PR #121, we now get nearly a 100% failure rate on message_filters. I tracked it down to "Timed out waiting for transform" exceptions from here https://github.com/ros2/geometry2/blob/fccfd986ad82208dc64d0f1d650dc87cf336cae7/tf2_ros/src/buffer.cpp#L257

I noticed message_filter sets a timeout of 0ns on waitForTransform. here https://github.com/ros2/geometry2/blob/fccfd986ad82208dc64d0f1d650dc87cf336cae7/tf2_ros/include/tf2_ros/message_filter.h#L327 and here https://github.com/ros2/geometry2/blob/fccfd986ad82208dc64d0f1d650dc87cf336cae7/tf2_ros/include/tf2_ros/message_filter.h#L352

0ns timeout is not enough time to receive the incoming transform data, so nearly every waitForTransform call times out. If I use a reasonable value for the timeout (200ms in this case) message filter starts working again.

It seems like this timeout should be a parameter to the message filter constructor. Thoughts?

crdelsey, 2019-08-06:
Or, is the intent that message_filter should never time out messages; instead, it just drops them when the queue is full? That's the impression I get looking at the interface.

jacobperron, 2019-08-06:
This is a bug. Addressed in #146.

jacobperron, 2019-08-06:
> It seems like this timeout should be a parameter to the message filter constructor. Thoughts?

This might make more sense then what I've done. I'll consider updating #146.

crdelsey, 2019-08-07:
#146 fixes the problem for me. Thanks.

closed 2019-08-07 by jacobperron