ros2/geometry2 #150: tf2::BufferCore fails to lookupTransform after clear

vinnamkim, 2019-08-20:
## Bug report

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - https://github.com/ros2/geometry2/commit/f282d2767e4d60d808be4a9cdf9a7019404b6e48
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
The following test is failed.
```
TEST(tf2_clear, LookUp_Static_Transfrom_Fail)
{
  tf2::BufferCore tfc;
  geometry_msgs::msg::TransformStamped st;
  st.header.frame_id = "foo";
  st.header.stamp = builtin_interfaces::msg::Time();
  st.header.stamp.sec = 1;
  st.header.stamp.nanosec = 0;
  st.child_frame_id = "bar";
  st.transform.rotation.w = 1;
  EXPECT_TRUE(tfc.setTransform(st, "authority1"));
  tfc.clear();
  EXPECT_TRUE(tfc.setTransform(st, "authority1", true));
  EXPECT_NO_THROW(
    auto trans = tfc.lookupTransform("foo", "bar", tf2::TimePoint(std::chrono::seconds(2)));
  );
}
```

In contrast, the following test is succeed.
```
TEST(tf2_clear, LookUp_Static_Transfrom_Succeed)
{
  tf2::BufferCore tfc;
  geometry_msgs::msg::TransformStamped st;
  st.header.frame_id = "foo";
  st.header.stamp = builtin_interfaces::msg::Time();
  st.header.stamp.sec = 1;
  st.header.stamp.nanosec = 0;
  st.child_frame_id = "bar";
  st.transform.rotation.w = 1;
  tfc.clear();
  EXPECT_TRUE(tfc.setTransform(st, "authority1", true));
  EXPECT_NO_THROW(
    auto trans = tfc.lookupTransform("foo", "bar", tf2::TimePoint(std::chrono::seconds(2)));
  );
}
```

#### Expected behavior
The first test should be succeed.

#### Actual behavior
It is failed.

#### Additional information
It seems there would be some error on clearing the buffer. Like the tests above, if we set non-static transform to the buffer firstly, then set the identical transform except it's static to the buffer, `lookupTransform` throws up an extrapolation exception:
```
C++ exception with description "Lookup would require extrapolation at time 2.00000, but only time 1.00000 is in the buffer, when looking up transform from frame [bar] to frame [foo]" thrown in the test body.
```
It seems a bug since a static transform should have no extrapolation issue like the second test above is passing.

vinnamkim, 2019-08-20:
It is occurred since `tf2::BufferCore` can't change a type of `TimeCacheInterfacePtr` in `frames_`. It seems there would be two approaches to the problem.

1. `tf2::BufferCore::clear()` removes `frames_` not just calling `TimeCacheInterfacePtr::clearList()`.
2. `tf2::BufferCore::setTransform()` overwrites a type of `TimeCacheInterfacePtr` with a current input(static or non-static).

Some discussions would be needed for whether `tf2::BufferCore` can change a type of `TimeCacheInterfacePtr` or not. Anyway, I made a PR(#151) which allows to overwrite the type.

clalancette, 2020-08-13:
I believe this was fixed by #151, so I'm going to close this.  If that's not the case, please feel free to reopen.

closed 2020-08-13 by clalancette