ros2/geometry2 #168: Style cleanup on tf2_py.cpp

clalancette, 2019-09-16:
After #99 is merged, there is a bunch of style cleanup that should be done on https://github.com/ros2/geometry2/blob/ros2/tf2_py/src/tf2_py.cpp .  We should do the following:

1.  Switch NULL -> nullptr
1.  Use consistent bracket style everywhere
1.  Enable the cpplint/uncrustify checks from ament to keep the style consistent in the future.

ahcorde, 2020-01-22:
PR https://github.com/ros2/geometry2/pull/222

ahcorde, 2020-02-18:
excluded `ament_copyright` and `ament_cpplint`

I have seen that bsd 3 was included, but I need to make changes (reformat the license) in the header to pass `ament_copyright` and `ament_cpplint`

https://github.com/ament/ament_lint/blob/master/ament_copyright/ament_copyright/template/bsd_3clause_header.txt

clalancette, 2020-08-13:
Fixed by #222 , so closing.

closed 2020-08-13 by clalancette