ros2/geometry2 #169: Split the tf2_ros package into tf2_ros and  tf2_ros_py

clalancette, 2019-09-16:
As discussed in the review for #99, the tf2_ros package currently has both the C++ and Python interface embedded into the same package.  This is problematic because tf2_py wants to depend on the tf2_ros C++ code, and be depended on by the tf2_ros Python code, leading to a circular dependency.  We currently break that dependency by copying code from tf2_ros into tf2_py, but that is not ideal.  Instead, we should break it up so that tf2_ros contains just the C++ code/interfaces, and introduce a new package tf2_ros_py that contains the Python code/interfaces.

jacobperron, 2020-07-22:
I think this was done in #210. Re-open if I'm mistaken.

closed 2020-07-22 by jacobperron

clalancette, 2020-07-22:
Yep, you are right, thanks for closing it out!