ros2/geometry2 #170: Change tf2_ros Python buffer client to a service?

clalancette, 2019-09-16:
In the port of tf2_ros Python to ROS 2 in #99, there was a TODO [added](https://github.com/ros2/geometry2/pull/99#pullrequestreview-273168135) asking why the buffer client is an action and not a service.  We should revisit this decision and switch it to a service (maybe an asynchronous one) if it makes sense.

jacobperron, 2019-09-18:
This question came up when I was porting the C++ buffer action client and server.
The answer I got was that actions have the benefit of being preempt-able. E.g. a client can request a transform and then later cancel the request.
That being said, if we implemented them with services, an implementer could simply ignore a late reply.

If we do decide to change to services, we should also update the C++ implementation for interoperability.

jacobperron, 2019-09-18:
I just noticed that there isn't an API exposed to cancel a goal in the C++ implementation :man_shrugging:

tfoote, 2019-09-19:
Yeah, for the basic api as it is the preemptability isn't super important. All of the existing API is designed to duplicate the core API which is synchronous.