ros2/geometry2 #174: Remove unused setup.py files

clalancette, 2019-09-26:
There are 4 setup.py files in this repository that look like they are no longer used:

* https://github.com/ros2/geometry2/blob/b2706c5cf3aa761112c15b7f816be89fa63a315e/tf2_kdl/setup.py
* https://github.com/ros2/geometry2/blob/b2706c5cf3aa761112c15b7f816be89fa63a315e/tf2_sensor_msgs/setup.py
* https://github.com/ros2/geometry2/blob/b2706c5cf3aa761112c15b7f816be89fa63a315e/tf2_ros/setup.py
* https://github.com/ros2/geometry2/blob/b2706c5cf3aa761112c15b7f816be89fa63a315e/tf2_geometry_msgs/setup.py

We should remove them as dead code.

dirk-thomas, 2019-09-26:
Wouldn't it have been similar effort to create a PR instead of this ticket?

clalancette, 2019-09-27:
> Wouldn't it have been similar effort to create a PR instead of this ticket?

I was busy doing releases, so no.  I'll get around to it when I can.

PR #190 (2019-11-19): Remove unused setup.py files
Closes  #174
closed 2019-11-19 by tfoote