ros2/geometry2 #176: tf2_geometry_msgs geometry_msgs to TF2 StampedTransform

SteveMacenski, 2019-09-27:
## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04
  - 
- Installation type: Dashing latest release, E master on hash below
  - 
- Version or commit hash: 4e519f3e77611a18f03398b6334e9cc8ccf49552
  - 
- DDS implementation: Fast RTPS
  - 
- Client library (if applicable): rclcpp
  - 

#### Steps to reproduce issue

try to call

```
geometry_msgs::msg::TransformStamped toMsg(const tf2::Stamped<tf2::Transform>& in)
```
Defined here: https://github.com/ros2/geometry2/blob/ros2/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L513

#### Expected behavior

I'm trying to convert a bunch of types around, and I've noticed a weird issue where all my other conversions work fine, except for this one. I get errors stating that this particular overload of `toMsg` doesnt exist. It works when I get out only a Transform, but not a stamped transform. I attempted both on Dashing from debians and on my ros2 master build, either recognize this and given it was written awhile ago I'm a bit confused why its not working.

#### Actual behavior
Typical compiler error saying it can't be found.  

```
/home/s.macenski/Documents/slam_ws/src/slam_toolbox/src/slam_toolbox_common.cpp: In member function ‘tf2::Stamped<tf2::Transform> slam_toolbox::SlamToolbox::setTransformFromPoses(const karto::Pose2&, const karto::Pose2&, const rclcpp::Time&, const bool&)’:
/home/s.macenski/Documents/slam_ws/src/slam_toolbox/src/slam_toolbox_common.cpp:398:45: error: no match for ‘operator=’ (operand types are ‘geometry_msgs::msg::TransformStamped {aka geometry_msgs::msg::TransformStamped_<std::allocator<void> >}’ and ‘geometry_msgs::msg::Transform {aka geometry_msgs::msg::Transform_<std::allocator<void> >}’)
     base_to_map_msg = tf2::toMsg(base_to_map);
```

#### Additional information

Specific set of code, while migrating Slam Toolbox to ROS2
```
  // Compute the map->odom transform
  tf2::Stamped<tf2::Transform> odom_to_map;
  tf2::Quaternion q(0.,0.,0.,1.0);
  q.setRPY(0., 0., corrected_pose.GetHeading());
  tf2::Stamped<tf2::Transform> base_to_map(
    tf2::Transform(q, tf2::Vector3(corrected_pose.GetX(),
    corrected_pose.GetY(), 0.0)).inverse(), tf2_ros::fromMsg(t), base_frame_);
  try
  {
    geometry_msgs::msg::TransformStamped base_to_map_msg, odom_to_map_msg;
    base_to_map_msg = tf2::toMsg(base_to_map);

    // this stuff works, obviously
    // base_to_map_msg.header.stamp = tf2_ros::toMsg(base_to_map.stamp_);
    // base_to_map_msg.header.frame_id = base_to_map.frame_id_;
    // base_to_map_msg.transform.translation.x = base_to_map.getOrigin().getX();
    // base_to_map_msg.transform.translation.y = base_to_map.getOrigin().getY();
    // base_to_map_msg.transform.translation.z = base_to_map.getOrigin().getZ();
    // base_to_map_msg.transform.rotation = tf2::toMsg(base_to_map.getRotation());

    odom_to_map_msg = tf_->transform(base_to_map_msg, odom_frame_);
    tf2::fromMsg(odom_to_map_msg, odom_to_map);
  }
  catch(tf2::TransformException & e)
  {
    RCLCPP_ERROR(get_logger(), "Transform from base_link to odom failed: %s",
      e.what());
    odom_to_map.setIdentity();
  }
```

And its counterpart, fromMsg, works fine between `tf2::stamped<tf2::transform>` and `geometry_msgs/transformstamped`

jacobperron, 2019-10-10:
The same error happens with the `toMsg` call for stamped quaternions. 

Since the `tf2::Stamped<T>` classes are defined as a subclass of `T`, the compiler is preferring the overloaded methods. E.g. that's why if you look closely at your error message it complaining about assigning a non-stamped type `geometry_msgs::msg::Transform` to a `geometry_msgs::msg::TransformStamped`:

    error: no match for ‘operator=’ (operand types are ‘geometry_msgs::msg::TransformStamped {aka geometry_msgs::msg::TransformStamped_<std::allocator<void> >}’ and ‘geometry_msgs::msg::Transform {aka geometry_msgs::msg::Transform_<std::allocator<void> >}’)

One workaround is to explicitly provide the template arguments:

    base_to_map_msg = tf2::toMsg<tf2::Stamped<tf2::Transform>, geometry_msgs::msg::TransformStamped>(base_to_map);

Here's a proposed fix, which changes the template specialization into an overload:  https://github.com/ros2/geometry2/pull/179

SteveMacenski, 2019-10-11:
Awesome! I was a little iffy on filing the bug at all since I was ***sure** it was something I did. Thanks for taking care of it, that all looks and sounds rational

sloretz, 2019-10-14:
It seems like the code used [by slam_toolbox in ROS 1 `tf2::convert(base_to_map, base_to_map_msg)`](https://github.com/SteveMacenski/slam_toolbox/blob/7dba9a7296dda2e9cec943b8c4873b9bf1b4e800/src/slam_toolbox_common.cpp#L352) might be fixed by #167.

SteveMacenski, 2019-10-14:
Looks like that PR's been sitting around for awhile. Would love to make it back the same but seems like both are issues worth addressing

closed 2019-10-19 by jacobperron

SteveMacenski, 2019-10-19:
Thanks @jacobperron

commit bed75c7b (2019-10-19): Remove toMsg template specialization for stamped transforms

This fixes #176 as demonstrated by the unit test compiling.
PR #179 (2019-10-19): Remove template specialization for toMsg functions
Resolves #176 

Before, the compiler was preferring the overloaded function:

https://github.com/ros2/geometry2/blob/71b054249cda9f1bbdbdc2f61bd518b998ce0378/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L397-L398

instead of the template specialization. I tried to change the overload into a template specialization, but then we run into the error that the compile can't deduce the second template parameter (ie. the return type).

I'm not sure if this is an ideal fix, or if instead we should go with template specialization (I'm [... cut]
commit 881362fc (2019-10-19): Remove template specialization for toMsg functions (#179)

* Add unit test for converting to/from stamped transform types

* Remove toMsg template specialization for stamped transforms

This fixes #176 as demonstrated by the unit test compiling.

* Fix test

* Remove toMsg template specialization for stamped quaternions

Similar to stamped transforms, this resolves a compilation issue when using the function.

* Expand test

* Remove last occurance of fromMsg template specialization

* Update tests to use tf2::convert