ros2/geometry2 #183: tf2_ros::StaticTransformBroadcaster does not keep last msg for later subscription

sharronliu, 2019-10-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - Dashing
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
In rclcpp node, send static TF using `tf2_ros::StaticTransformBroadcaster::sendTransform()`. The function was invoked only once, since the TF is unchanged. The observation is the msg was not received by a later subscription. This behavior differs from ROS1.
Try to instanciate the rclcpp node with below QoS profile, problem still observed.
- rclcpp::Qos(rclcpp::KeepLast(1))
- rclcpp::Qos(rclcpp::KeepLast(10))
- rclcpp::Qos(rclcpp::KeepAll())

Please can you suggest any other parameters may lead to the problematic result?

#### Expected behavior
The static msg was kept for later subscription

#### Actual behavior
The static msg was not received in later subscription

clalancette, 2019-10-21:
It is kept, but the naming in DDS (and hence ROS 2) is different.  In ROS 2, a "latched" topic is called "transient local".  If the publisher is a "transient local", then it will hold a copy for any subscriber that comes in later.  However, the subscriber must also be "transient local" for this to work.

If you use the stock TF2 clients, they have the appropriate QoS settings for this to work.  If you are subscribing to `tf_static` on your own, you'll need to set the transient_local QoS setting.

I'm going to close this out assuming that is what is going on here, but feel free to keep commenting or re-open if this doesn't work for you.

closed 2019-10-21 by clalancette

sloretz, 2019-10-21:
@sharronliu I noticed you're using `Dashing`. The static listener and broadcaster use transient local durability as of #160, so it should behave as you expect in `Eloquent`.