ros2/geometry2 #185: lookupTransform blocks forever

maxlein, 2019-10-22:
When I lookup transforms with a timeout parameter, the function blocks forever when there exists no transform ...  
Without timeout parameter, I get an immediate exception that the frame doesn't exist

## Bug report

**Required Info:**

- Operating System:
Ubuntu 18.04
- Installation type:
binaries
- Version or commit hash:
Dashing
- DDS implementation:
opensplice
- Client library (if applicable):
rclcpp

#### Steps to reproduce issue

`auto tmpTransform = tf->lookupTransform("base_link", tf2::TimePointZero, "base_link", tf2::TimePointZero, "non_existing_frame", tf2::Duration(100ms) ); 
`       

This is working correctly:  

`auto tmpTransform = tf->lookupTransform("base_link", tf2::TimePointZero, "base_link", tf2::TimePointZero, "non_existing_frame" );
`      

#### Expected behavior
terminate called after throwing an instance of 'tf2::LookupException'
  what():  "non_existing_frame" passed to lookupTransform argument fixed_frame does not exist. 

#### Actual behavior
System blocks forever, no exception

nuclearsandwich, 2019-12-06:
@maxlein I wasn't able to reproduce a hang using OpenSplice from debs. Can you check that you're still hitting this bug, and if so provide a complete example package which demonstrates it to aid my investigations?

clalancette, 2020-10-21:
Closing due to inactivity.  Feel free to reopen if you'd still like to investigate this issue.

closed 2020-10-21 by clalancette

Lannister-Xiaolin, 2022-04-27:
I had meet the same problem in foxy, when set timeout param, I switch to galactic version or delete timeout param, the block out problem disapeared！！

[image]