ros2/geometry2 #188: Unable to import tf2_ros

RoboticsYY, 2019-10-23:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - from source 
- Version or commit hash:
  - c3cabae25308c35e5f7d3e6ee879aa5b052dc711
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

- Install ROS2 [Dashing](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Install-Debians/)
- Build ros2/geometry2
```shell
source /opt/ros/dashing/setup.bash
mkdir -p ~/ros2_ws/src && cd ~/ws_ros2_ws/src
git clone https://github.com/ros2/geometry2.git
cd geometry2 && git checkout c3cabae25308c35e5f7d3e6ee879aa5b052dc711
cd ../.. 
colcon build
source install/setup.bash
```
- Import `tf2_ros` in any python code. 

#### Actual behavior

Came into the error: `No module named _tf2_py`

#### Additional information

`_tf2_py.cpython-36m-x86_64-linux-gnu.so` and `__init__.py` existing in the `~/ros2_ws/install/tf2_py/lib/python3.6/site-packages/tf2_py` folder.

This effort targets using the tf2_ros in python, which is enabled in the #99.

clalancette, 2019-10-23:
Unfortunately, the tf2_ros python port is not going to work with Dashing (or, at least, it's not going to work with Dashing without significant additional work).  That's because the port uses some APIs from Eloquent.  Actually, I'm surprised it compiled at all for you, since I thought we were using APIs that didn't exist in Dashing.  @sloretz , thoughts?  I'm going to close this out as this isn't a currently supported use-case, but it would be good to know why it compiles.

closed 2019-10-23 by clalancette

sloretz, 2019-10-23:
> @sloretz , thoughts? I'm going to close this out as this isn't a currently supported use-case, but it would be good to know why it compiles.

The python port is not backported to dashing because it uses https://github.com/ros2/rclpy/pull/412 and the Python static transform broadcaster and listener use `transient_local` durability following https://github.com/ros2/geometry2/pull/160 . Both of those would show up as test or runtime issues rather than compile errors. I don't know what other changes have been added to geometry2 since Dashing though.

RoboticsYY, 2019-10-23:
Thanks for the answers. Looking forward to the tf2_ros python. Most of our ros1 python projects need this to do the porting. Currently I take a substitution with a cpp service.