ros2/geometry2 #194: Python call callback with transform data when it becomes available

sloretz, 2019-11-15:
## Feature

It would be convenient if there was an API that returned transform information in a callback.

#### Feature description

It looks like C++ has an overload of `waits_for_transform` that calls the callback with a future containing the transform.
https://github.com/ros2/geometry2/blob/71f823e90b699c9535ce682db777f7bc88cdfbcd/tf2_ros/include/tf2_ros/buffer.h#L214-L216

#### Implementation considerations

 It should be possible using the callbacks on the future returned by `wait_for_transform_async`, however the returned future is not associated with an executor so done callbacks are never called (ros2/rclpy#460).

https://github.com/ros2/geometry2/blob/71f823e90b699c9535ce682db777f7bc88cdfbcd/tf2_ros/src/tf2_ros/buffer.py#L217-L226

However, even if the done callbacks were called the result on future doesn't contain any info about the transform. It should probably have that.

allenh1, 2019-12-06:
@sloretz would you mind if I looked into doing this?

sloretz, 2019-12-06:
> @sloretz would you mind if I looked into doing this?

@allenh1 please do, and thank you!

allenh1, 2020-01-12:
> It should be possible using the callbacks on the future returned by wait_for_transform_async, however the returned future is not associated with an executor so done callbacks are never called (ros2/rclpy#460).

@sloretz since the issue you mentioned is closed now, I'm assuming this can be done as you were saying by using `wait_for_transform_async`, right? I'll send a PR in with that assumption and we can talk further if that wasn't a good one.

allenh1, 2021-01-30:
I've removed myself from this because I don't have the time to do this before the galactic release. Anyone who feels led to should feel free to pick up this issue. Otherwise, I might have time in a while to revisit this.

PR #664 (2024-04-04): Transform Data Callback Python
This PR is meant to resolve #194 as of right now I'm working off of PR #215. We need to change `_wait_for_transform_async`, to return the transform data similar to the [C++ version](https://github.com/ros2/geometry2/blob/71f823e90b699c9535ce682db777f7bc88cdfbcd/tf2_ros/src/buffer.cpp#L195-L260) returning a `TransformStampedFuture`.
closed 2024-04-04 by ahcorde