ros2/geometry2 #204: Increase timeout_padding for BufferClient tests to support build farm armhf tests

piraka9011, 2019-12-11:
## Bug report

- Operating System:
  - Ubuntu 18.04 (Build farm)
- Installation type:
  - Source
- Version or commit hash:
  - `master`
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

Run the build farm tests:
`colcon test --packages-select tf2_ros`

#### Expected behavior

Tests Pass

#### Actual behavior

[Tests fail](https://ci.ros2.org/view/nightly/job/nightly_linux-armhf_debug/308/testReport/junit/tf2_ros.src.ros2.geometry2.tf2_ros.test.py_test.test_buffer_client/TestBufferClient/test_lookup_transform_fail/) with `tf2.TimeoutException`. 

#### Additional information

[Failed Build Example](https://github.com/ros2/geometry2/blob/88cb824a1a069cbb3e4bd2eb9c10b626b00fe0ac/tf2_ros/test/py_test/test_buffer_client.py#L190)

Potential solutions:
- Increasing the `timeout_padding` for the `BufferClient`.
- Allowing the test to pass if a `TimeoutException` is raised (not recommended).

Would prefer to get some feedback on how to approach this before opening a PR.

piraka9011, 2019-12-12:
I tried increasing the `timeout_padding` of the `BufferClient` and it looks like that does not fix the test. It seems this is a deeper issue in the buffer client.

When the `BufferClient` [processes a goal](https://github.com/ros2/geometry2/blob/88cb824a1a069cbb3e4bd2eb9c10b626b00fe0ac/tf2_ros/src/tf2_ros/buffer_client.py#L159) it immediately checks if the server is ready without ever calling a `wait_for_server()` anywhere.

I don't know if it's fine to block the initialization of the client but I suggest doing the following:

```python
class BufferClient(tf2_ros.BufferInterface):
   ...
   self.action_client = ActionClient(node, LookupTransform, action_name=ns)
   self.action_client.wait_for_server(self.timeout_padding)
```

@tfoote @wjwwood thoughts?

tfoote, 2019-12-17:
You added the wait_for_server just before this?   https://github.com/ros2/geometry2/blob/88cb824a1a069cbb3e4bd2eb9c10b626b00fe0ac/tf2_ros/test/py_test/test_buffer_client.py#L194

We could also add a wait and then assert there that the client is connected at that time to close that possibility of a startup race condition.

We shouldn't add the wait in the constructor. Calling code should never make the assumption that the server is guarenteed there. For example, it might go away after construction and you should be robust that's why the wait in the test makes more sense.

I also note that the `timeout_padding` in the test is 0.0 so if you used that it probably didn't make much difference.

piraka9011, 2019-12-17:
> You added the wait_for_server just before this?

Yes

> We could also add a wait and then assert there that the client is connected at that time to close that possibility of a startup race condition.

It doesn't look like the client ever connects (waited for more than a minute), unless it takes a really long time to connect....

> I also note that the timeout_padding in the test is 0.0 so if you used that it probably didn't make much difference.

I increased that when testing locally and the test still failed.

tfoote, 2020-01-02:
Hmm this now looks like an issue with the action client initializing then. If you can isolate this down to a simplified example we should refile it onto the actions.

clalancette, 2024-12-04:
We are no longer building for armhf, and this issue is very stale, so I'm closing it for now.  Feel free to reopen if you want to keep debugging it.

closed 2024-12-04 by clalancette