ros2/geometry2 #212: tf2_ros test_buffer_client.py test is not working in CI

ahcorde, 2020-01-10:
tf2_ros test_buffer_client.py test is not working in CI.

Weird behaviour, test is fully working in the local machine but inside the CI the test is failing.

ivanpauno, 2020-01-30:
@ahcorde can you add a link to the failing CI test?
Can this be assigned to you?

ahcorde, 2020-01-30:
The complete log is here: https://ci.ros2.org/job/ci_linux/8938/console

```
15:02:43       Start  8: tf2_ros_test_time_reset
15:02:43 
15:02:43 8: Test command: /home/jenkins-agent/workspace/ci_linux/venv/bin/python3 "-u" "/home/jenkins-agent/workspace/ci_linux/ws/install/ament_cmake_test/share/ament_cmake_test/cmake/run_test.py" "/home/jenkins-agent/workspace/ci_linux/ws/build/tf2_ros/test_results/tf2_ros/tf2_ros_test_time_reset.gtest.xml" "--package-name" "tf2_ros" "--output-file" "/home/jenkins-agent/workspace/ci_linux/ws/build/tf2_ros/ament_cmake_gtest/tf2_ros_test_time_reset.txt" "--command" "/home/jenkins-agent/workspace/ci_linux/ws/build/tf2_ros/tf2_ros_test_time_reset" "--gtest_output=xml:/home/jenkins-agent/workspace/ci_linux/ws/build/tf2_ros/test_results/tf2_ros/tf2_ros_test_time_reset.gtest.xml"
15:02:43 8: Test timeout computed to be: 60
15:02:43 8: -- run_test.py: invoking following command in '/home/jenkins-agent/workspace/ci_linux/ws/build/tf2_ros':
15:02:43 8:  - /home/jenkins-agent/workspace/ci_linux/ws/build/tf2_ros/tf2_ros_test_time_reset --gtest_output=xml:/home/jenkins-agent/workspace/ci_linux/ws/build/tf2_ros/test_results/tf2_ros/tf2_ros_test_time_reset.gtest.xml
15:02:43 8: [==========] Running 1 test from 1 test case.
15:02:43 8: [----------] Global test environment set-up.
15:02:43 8: [----------] 1 test from tf2_ros_time_reset_test
15:02:43 8: [ RUN      ] tf2_ros_time_reset_test.time_backwards
15:02:43 8: Warning: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
15:02:43 8:          at line 133 in /home/jenkins-agent/workspace/ci_linux/ws/src/ros2/geometry2/tf2/src/buffer_core.cpp
15:02:44 8: Warning: Invalid frame ID "foo" passed to canTransform argument target_frame - frame does not exist
15:02:44 8:          at line 133 in /home/jenkins-agent/workspace/ci_linux/ws/src/ros2/geometry2/tf2/src/buffer_core.cpp
15:02:44 8: /home/jenkins-agent/workspace/ci_linux/ws/src/ros2/geometry2/tf2_ros/test/time_reset_test.cpp:85: Failure
15:02:44 8: Value of: buffer.canTransform("foo", "bar", rclcpp::Time(101, 0))
15:02:44 8:   Actual: false
15:02:44 8: Expected: true
15:02:44 8: [  FAILED  ] tf2_ros_time_reset_test.time_backwards (1015 ms)
15:02:44 8: [----------] 1 test from tf2_ros_time_reset_test (1015 ms total)
15:02:44 8: 
15:02:44 8: [----------] Global test environment tear-down
15:02:44 8: [==========] 1 test from 1 test case ran. (1015 ms total)
15:02:44 8: [  PASSED  ] 0 tests.
15:02:44 8: [  FAILED  ] 1 test, listed below:
15:02:44 8: [  FAILED  ] tf2_ros_time_reset_test.time_backwards
15:02:44 8: 
15:02:44 8:  1 FAILED TEST
15:02:44 8: -- run_test.py: return code 1
15:02:44 8: -- run_test.py: verify result file '/home/jenkins-agent/workspace/ci_linux/ws/build/tf2_ros/test_results/tf2_ros/tf2_ros_test_time_reset.gtest.xml'
15:02:44  8/10 Test  #8: tf2_ros_test_time_reset .....................***Failed    1.13 sec
 ``` 

Sure, you can assign me this task

ivanpauno, 2020-01-30:
> Sure, you can assign me this task

Thanks!

clalancette, 2020-08-13:
Looking at the latest [nightlies](https://ci.ros2.org/view/nightly/job/nightly_linux_release/1639/), it doesn't look like this is a problem anymore.  I'm going to close this out, but feel free to reopen if you think it is still a problem.

closed 2020-08-13 by clalancette