ros2/geometry2 #219: [TF] Slash or no slash, that is the question

maxlein, 2020-01-19:
My frames look like this:  

```
Frames:
Frame: /camera_link, published by <no authority available>, Average Delay: 0.0237926, Max Delay: 0.979575
Frame: /scan_link, published by <no authority available>, Average Delay: 0.0238071, Max Delay: 0.979573
Frame: base_link, published by <no authority available>, Average Delay: 0.00129464, Max Delay: 0.00788069
Frame: camera_optical_link, published by <no authority available>, Average Delay: 1629.37, Max Delay: 1629.37
Frame: odom, published by <no authority available>, Average Delay: -0.969808, Max Delay: 0
```  

I just use default ROS tools/ways of publishing tf's, so nothing special.

But robot_state_publisher is adding a slash to tf frames.  
And I think the rule is not to have a leading slash before frames.  
A tf publisher in the driver for example doesn't add a slash.  

Publisher is launched like this ( doesn't matter if I add namespace or not ):

Node(package='robot_state_publisher',
             node_namespace=LaunchConfiguration('namespace'),
             node_name='robot_state_publisher',
             node_executable='robot_state_publisher',
             arguments=[LaunchConfiguration('urdf')],
             output='screen'),

## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - eloquent
- DDS implementation:
  - cyclone dds

#### Steps to reproduce issue
Run robot state publisher with some robot urdf and look at tf monitor.

#### Expected behavior
No leading slashes  

#### Actual behavior
Leading slashes

tfoote, 2020-01-22:
There should not be a slash. Please file this against robot_state_publisher.

clalancette, 2020-01-22:
I'm moving this over to robot_state_publisher, but I'm not convinced it is a problem in code.  In the end, robot_state_publisher just parses the URDF and gets the frame names from that; it doesn't add any slashes on its own (I don't think).  What does your URDF look like?

clalancette, 2020-01-22:
Oh, actually, I can't transfer issues across organizations.  So I'll close this out; I'd suggest posting to answers.ros.org for a solution, and if you still think it is a bug after that, opening a bug against https://github.com/ros/robot_state_publisher.

closed 2020-01-22 by clalancette