ros2/geometry2 #228: tf2 test: Differences between ci.ros2.org and build.ros2.org

ahcorde, 2020-01-31:
Similar to this other issue https://github.com/ros2/geometry2/issues/212. there are some tests which are working in [ci.ros2.org](http://ci.ros2.org) and not in [build.ros2.org](http://build.ros2.org)

In the CI the tests are passing: 

```
14:17:05 7: [==========] Running 3 tests from 1 test case.
14:17:05 7: [----------] Global test environment set-up.
14:17:05 7: [----------] 3 tests from tf2_test_transform_broadcaster
14:17:05 7: [ RUN      ] tf2_test_transform_broadcaster.transform_broadcaster_rclcpp_node
14:17:05 7: [       OK ] tf2_test_transform_broadcaster.transform_broadcaster_rclcpp_node (10 ms)
14:17:05 7: [ RUN      ] tf2_test_transform_broadcaster.transform_broadcaster_custom_rclcpp_node
14:17:05 7: [       OK ] tf2_test_transform_broadcaster.transform_broadcaster_custom_rclcpp_node (3 ms)
14:17:05 7: [ RUN      ] tf2_test_transform_broadcaster.transform_broadcaster_as_member
14:17:05 7: [       OK ] tf2_test_transform_broadcaster.transform_broadcaster_as_member (3 ms)
14:17:05 7: [----------] 3 tests from tf2_test_transform_broadcaster (16 ms total)
14:17:05 7: 
14:17:05 7: [----------] Global test environment tear-down
14:17:05 7: [==========] 3 tests from 1 test case ran. (16 ms total)
14:17:05 7: [  PASSED  ] 3 tests.
14:17:05 7: -- run_test.py: return code 0
14:17:05 7: -- run_test.py: inject classname prefix into gtest result file '/home/jenkins-
```

But in build.ros2.org a timeout has raised

```
23:44:14 7: Test timeout computed to be: 60
23:44:14 7: -- run_test.py: invoking following command in '/tmp/ws/src/geometry2/tf2_ros':
23:44:14 7:  - /tmp/ws/build_isolated/tf2_ros/tf2_ros_test_transform_broadcaster --gtest_output=xml:/tmp/ws/test_results/tf2_ros/tf2_ros_test_transform_broadcaster.gtest.xml
23:44:14 7: [==========] Running 3 tests from 1 test case.
23:44:14 7: [----------] Global test environment set-up.
23:44:14 7: [----------] 3 tests from tf2_test_transform_broadcaster
23:44:14 7: [ RUN      ] tf2_test_transform_broadcaster.transform_broadcaster_rclcpp_node
23:44:14 7: [       OK ] tf2_test_transform_broadcaster.transform_broadcaster_rclcpp_node (4 ms)
23:44:14 7: [ RUN      ] tf2_test_transform_broadcaster.transform_broadcaster_custom_rclcpp_node
23:44:14 7: [       OK ] tf2_test_transform_broadcaster.transform_broadcaster_custom_rclcpp_node (2 ms)
23:44:14 7: [ RUN      ] tf2_test_transform_broadcaster.transform_broadcaster_as_member
23:45:14  7/10 Test  #7: tf2_ros_test_transform_broadcaster ..........***Timeout  60.00 sec
23:45:14 test 8
23:45:14       Start  8: tf2_ros_test_time_reset
```

ahcorde, 2020-01-31:
CI job https://ci.ros2.org/job/ci_linux/9134/
Build job: http://build.ros2.org/job/Epr__geometry2__ubuntu_bionic_amd64/88/console

clalancette, 2020-08-13:
This looks like it is still somewhat a problem in Eloquent PR jobs, but the Rolling PR jobs haven't failed in over a week: http://build.ros2.org/view/Rpr/job/Rpr__geometry2__ubuntu_focal_amd64/ .  I'm going to close this out, but feel free to reopen if you think this is still a problem.

closed 2020-08-13 by clalancette