ros2/geometry2 #236: transformPose is not supported in python?

ipa-hsd, 2020-02-12:
Seems like `transformPose` method is still not implemented in python for Eloquent.
For ROS(1), it was implemented in [listener.py](https://github.com/ros/geometry/blob/melodic-devel/tf/src/tf/listener.py#L286) (which uses `tf2_ros` under the hood) which uses [transformations.py](https://github.com/ros/geometry/blob/melodic-devel/tf/src/tf/transformations.py) (which mainly depends on `numpy`). How is it planned for ROS2 if I have to contribute?

clalancette, 2020-02-12:
`transformPose` is a legacy API in `tf`, which has long been deprecated in favor of `tf2`.  In ROS 2, we've removed `tf` completely and just have `tf2`.

However, if I understand properly, the functionality you are looking for exists in the `tf2` python layer.  Basically you need to create a BufferInterface object, and then call `transform` on it, which I think will do what you are looking for.  See https://github.com/ros2/geometry2/blob/ros2/tf2_ros/doc/index.rst for a bit more additional information.

I'm going to close this issue.  If you still have questions about how to use this, please open a question at https://answers.ros.org, since we like to keep all questions and answers at a single place.

closed 2020-02-12 by clalancette

ipa-hsd, 2020-02-13:
Sorry but seems like the necessary package is not ported yet?  
https://github.com/ros2/geometry2/blob/eloquent/tf2_geometry_msgs/CMakeLists.txt#L31

ipa-hsd, 2020-02-14:
For your reference: https://answers.ros.org/question/344028/transform-pose-in-tf2/