ros2/geometry2 #237: ROS2 TF2: Severing the fork

SteveMacenski, 2020-02-13:
Hi,

This may be a bit of a weird request. I'm someone that occasionally uses the GitHub search functionality to find function signatures and quickly track down implementations. With Forks in GitHub, this functionality is disabled and only searches the parent repository. Given that there are some changes, it would be nice if that functionality was restored. 

If you email the GitHub folks, they'll sever a fork for you into its same repo, same name, and allows this feature to come back. I did this with [slam toolbox](https://github.com/SteveMacenski/slam_toolbox) which originally started as a fork of slam karto. 

Also given that this is long term probably not going to be merged back in with ROS1's geometry2 repo, I think that long term this should become a stand-alone repository, since ros/geometry2 will be long dead in 10 years.

clalancette, 2020-03-02:
For what it's worth, I agree with this; they are now separate repositories.  So I'm happy to sever them.  I'd like to get approval from Tully before doing it, however.  @tfoote , thoughts?

SteveMacenski, 2020-03-02:
FYI, this is the place to do it: https://support.github.com/contact

The right words they look for are "detach fork" and they'll want to know what name you want it under (the same).

tfoote, 2020-03-25:
The codebase has definitely diverged. Optimally we'd merge back as in the end there will only be one maintained version and having an archived older version will lead to confusion but there's enough outstanding prs and issues that switching will be painful too so probably just detaching and maintaining the status quo makes sense.

clalancette, 2020-03-26:
All right, thanks for the feedback @tfoote .  I'll go ahead and open a ticket with Github.

clalancette, 2020-03-26:
This is now done.  I actually don't know if it is possible to link to a Github support ticket, but for reference this is ticket 615982 .  Thanks @SteveMacenski for bringing it up.

dirk-thomas, 2020-04-20:
@clalancette Please follow up on clarifying the branches in this repo. Currently there is a `master` branch as well as a `ros2` branch. One of them should be removed imo.

dirk-thomas, 2020-06-11:
@clalancette Friendly ping.

clalancette, 2020-06-15:
After discussing with @tfoote , we decided to keep the `ros2` branch as the main development branch, and get rid of the `master` branch.  I've now done that, along with removing a bunch of branches that either didn't pertain to ROS 2 (like groovy-devel, etc), or that were stale.

closed 2020-06-15 by clalancette