ros2/geometry2 #244: Seg fault constructing tf2_ros::BufferClient with empty pointer

ScottMcMichael, 2020-03-17:
## Bug report

**Required Info:**

- Operating System:  Ubuntu 18.04
- Installation type: Binary
- Version or commit hash: Build 1786

#### Steps to reproduce issue

rclcpp::Node::SharedPtr nh;
tf2_ros::BufferClient client(nh, "client");

#### Expected behavior

Some sort of error message

#### Actual behavior

Seg fault

hidmic, 2020-03-30:
This problem isn't actually related to `tf2_ros` but a result of an incomplete `rclcpp` API and it not checking for node nullity consistently. See https://github.com/ros2/rclcpp/pull/1035. CC @Karsten1987 .

hidmic, 2020-04-22:
@ScottMcMichael with https://github.com/ros2/rclcpp/pull/1069 in, this is now fixed. Thanks for reporting.

closed 2020-04-22 by hidmic