ros2/geometry2 #252: Tracking issue for windows test failures

tfoote, 2020-04-17:
This is an issue to track windows test failures.

MessageFilter.tolerance test is failing with: 
`2: unknown file: error: C++ exception with description "No such file or directory" thrown in the test body.`

https://ci.ros2.org/job/ci_windows-metal/9/testReport/

https://ci.ros2.org/job/ci_windows-metal/9/testReport/(root)/projectroot/test_message_filter/
<details>

```
2: Test command: C:\Python37\python.exe "-u" "C:/J/workspace/ci_windows-metal/ws/install/share/ament_cmake_test/cmake/run_test.py" "C:/J/workspace/ci_windows-metal/ws/build/test_tf2/test_results/test_tf2/test_message_filter.gtest.xml" "--package-name" "test_tf2" "--output-file" "C:/J/workspace/ci_windows-metal/ws/build/test_tf2/ament_cmake_gtest/test_message_filter.txt" "--command" "C:/J/workspace/ci_windows-metal/ws/build/test_tf2/Release/test_message_filter.exe" "--gtest_output=xml:C:/J/workspace/ci_windows-metal/ws/build/test_tf2/test_results/test_tf2/test_message_filter.gtest.xml"
2: Test timeout computed to be: 60
2: -- run_test.py: invoking following command in 'C:\J\workspace\ci_windows-metal\ws\build\test_tf2':
2:  - C:/J/workspace/ci_windows-metal/ws/build/test_tf2/Release/test_message_filter.exe --gtest_output=xml:C:/J/workspace/ci_windows-metal/ws/build/test_tf2/test_results/test_tf2/test_message_filter.gtest.xml
2: [==========] Running 7 tests from 1 test case.
2: [----------] Global test environment set-up.
2: [----------] 7 tests from MessageFilter
2: [ RUN      ] MessageFilter.noTransforms
2: [WARN] [1587078890.064792000] [tf2_ros_message_filter]: This rmw implementation does not support ON_OFFERED_INCOMPATIBLE_QOS events, you will not be notified when Publishers offer an incompatible QoS profile to Subscriptions on the same topic.
2: [WARN] [1587078890.065187500] [tf2_ros_message_filter]: This rmw implementation does not support ON_REQUESTED_INCOMPATIBLE_QOS events, you will not be notified when Subscriptions request an incompatible QoS profile from Publishers on the same topic.
2: [       OK ] MessageFilter.noTransforms (17 ms)
2: [ RUN      ] MessageFilter.noTransformsSameFrame
2: [       OK ] MessageFilter.noTransformsSameFrame (9 ms)
2: [ RUN      ] MessageFilter.preexistingTransforms
2: [       OK ] MessageFilter.preexistingTransforms (9 ms)
2: [ RUN      ] MessageFilter.postTransforms
2: [       OK ] MessageFilter.postTransforms (9 ms)
2: [ RUN      ] MessageFilter.setTargetFrame
2: [       OK ] MessageFilter.setTargetFrame (9 ms)
2: [ RUN      ] MessageFilter.multipleTargetFrames
2: [       OK ] MessageFilter.multipleTargetFrames (9 ms)
2: [ RUN      ] MessageFilter.tolerance
2: unknown file: error: C++ exception with description "No such file or directory" thrown in the test body.
2: [  FAILED  ] MessageFilter.tolerance (9 ms)
2: [----------] 7 tests from MessageFilter (71 ms total)
2: 
2: [----------] Global test environment tear-down
2: [==========] 7 tests from 1 test case ran. (71 ms total)
2: [  PASSED  ] 6 tests.
2: [  FAILED  ] 1 test, listed below:
2: [  FAILED  ] MessageFilter.tolerance
```

</details>

Also buffer_core_test is failing with what seems to be another file IO issue but different.

https://ci.ros2.org/job/ci_windows-metal/9/testReport/(root)/projectroot/buffer_core_test/

`unknown file: error: C++ exception with description "File exists" thrown in the test body.`

<details>

```

[----------] 3 tests from BufferCore_transformableCallbacks
[ RUN      ] BufferCore_transformableCallbacks.alreadyTransformable
[       OK ] BufferCore_transformableCallbacks.alreadyTransformable (0 ms)
[ RUN      ] BufferCore_transformableCallbacks.waitForNewTransform
unknown file: error: C++ exception with description "File exists" thrown in the test body.
[  FAILED  ] BufferCore_transformableCallbacks.waitForNewTransform (1 ms)
[ RUN      ] BufferCore_transformableCallbacks.waitForOldTransform
unknown file: error: C++ exception with description "File exists" thrown in the test body.
[  FAILED  ] BufferCore_transformableCallbacks.waitForOldTransform (0 ms)
[----------] 3 tests from BufferCore_transformableCallbacks (1 ms total)
```
</details

brawner, 2020-04-17:
buffer_core_test resolved by: https://github.com/ros2/geometry2/pull/253

brawner, 2020-04-17:
Has the issue about TF stderr printouts been ticketed somewhere for Windows? The buffer core test causes MSBuild warnings. Whatever jenkins is doing to find msbuild warnings should restrict things to build output right?

clalancette, 2020-08-13:
I believe all of the issues identified here are now resolved, so I'm going to close this out.  Feel free to reopen if I'm in error.

closed 2020-08-13 by clalancette