ros2/geometry2 #261: tf2_ros::TransformBroadcaster linker error

AndreasAZiegler, 2020-04-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18 (Bionic)
- Installation type:
  - binaries
- Version or commit hash:
  - eloquent
- DDS implementation:
  - RTI Connex
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)
project(transform-broadcaster)

# Default to C99
if(NOT CMAKE_C_STANDARD)
  set(CMAKE_C_STANDARD 99)
endif()

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic -ggdb3)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
find_package(rclcpp REQUIRED)
find_package(geometry_msgs REQUIRED)
find_package(tf2_ros REQUIRED)
find_package(tf2_msgs REQUIRED)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()

add_executable(transform-broadcaster src/transform-broadcaster.cpp)
ament_target_dependencies(transform-broadcaster rclcpp geometry_msgs)

install(TARGETS
  transform-broadcaster
  DESTINATION lib/${PROJECT_NAME})
```
transform-broadcaster.cpp
```
#include <chrono>
#include <memory>

#include "geometry_msgs/msg/transform_stamped.hpp"
#include "rclcpp/rclcpp.hpp"
#include "tf2_ros/transform_broadcaster.h"
#include <tf2_msgs/msg/tf_message.hpp>

using std::placeholders::_1;

class Broadcaster : public rclcpp::Node
{
public:
  Broadcaster() : Node("transform_broadcaster"), transform_broadcaster_{nullptr}
  {
    transform_broadcaster_ 
      = std::make_shared<tf2_ros::TransformBroadcaster>(this->shared_from_this());
    subscription_ = this->create_subscription<geometry_msgs::msg::TransformStamped>(
      "rcv_transform", 10, std::bind(&Broadcaster::transformCallback, this, _1));

    RCLCPP_WARN(this->get_logger(), "Node is up and running!");
  }

private:
  void transformCallback(const geometry_msgs::msg::TransformStamped::SharedPtr msg)
  {
    const geometry_msgs::msg::TransformStamped transform_stamped = *msg;
    transform_broadcaster_->sendTransform(transform_stamped);
  }

  rclcpp::Subscription<geometry_msgs::msg::TransformStamped>::SharedPtr subscription_;
  std::shared_ptr<tf2_ros::TransformBroadcaster> transform_broadcaster_;
};

int main(int argc, char *argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<Broadcaster>());
  rclcpp::shutdown();
  return 0;
}
```
Try to build it with ```colcon build --symlink-install```

#### Expected behavior
Build suceeds

#### Actual behavior
```
--- stderr: transform-broadcaster                              
CMakeFiles/transform-broadcaster.dir/src/transform-broadcaster.cpp.o: In function `Broadcaster::transformCallback(std::shared_ptr<geometry_msgs::msg::TransformStamped_<std::allocator<void> > >)':
/home/andreasziegler/mt-robot/routing_service_test_setup/ros2_ws/transform-broadcaster/src/transform-broadcaster.cpp:28: undefined reference to `tf2_ros::TransformBroadcaster::sendTransform(geometry_msgs::msg::TransformStamped_<std::allocator<void> > const&)'
CMakeFiles/transform-broadcaster.dir/src/transform-broadcaster.cpp.o: In function `rclcpp::Publisher<tf2_msgs::msg::TFMessage_<std::allocator<void> >, std::allocator<void> >::Publisher(rclcpp::node_interfaces::NodeBaseInterface*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rclcpp::QoS const&, rclcpp::PublisherOptionsWithAllocator<std::allocator<void> > const&)':
/opt/ros/eloquent/include/rclcpp/publisher.hpp:70: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<tf2_msgs::msg::TFMessage_<std::allocator<void> > >()'
collect2: error: ld returned 1 exit status
make[2]: *** [transform-broadcaster] Error 1
make[1]: *** [CMakeFiles/transform-broadcaster.dir/all] Error 2
make: *** [all] Error 2
---
Failed   <<< transform-broadcaster	[ Exited with code 2 ]
```

hidmic, 2020-05-07:
@AndreasAZiegler I'd think `tf2_ros` should be included in the `ament_target_dependencies()` call.

AndreasAZiegler, 2020-06-09:
@hidmic Thanks, that indeed solve this problem. Now, however I get a
```
terminate called after throwing an instance of 'std::bad_weak_ptr'
  what():  bad_weak_ptr
```
from ```rclcpp::spin(std::make_shared<Broadcaster>());```

closed 2020-06-09 by AndreasAZiegler