ros2/geometry2 #275: Add :type: and :rtype: annotations to tf2_ros_py code

clalancette, 2020-06-11:
This is a follow-up ticket to https://github.com/ros2/geometry2/pull/210 .  There, we split the tf2_ros python code out of the main tf2_ros code and into its own package.  While reviewing that code, @ahcorde noticed that the docblocks for each method are missing the `:type:` and `:rtype` annotations.  These would be nice to document for a little better documentation.

surfertas, 2020-08-29:
@clalancette @ahcorde Hi, I can work on this, but wanted to see if the preference was to use `:type:` and `:rtype` annotations, or use type hints as what is seen at [rclpy/node.py](https://github.com/ros2/rclpy/blob/master/rclpy/rclpy/node.py) .

clalancette, 2020-08-31:
@surfertas Thanks for the offer!

You make a good point; it is probably a good idea to use the type hints instead to be compatible with what rclpy is doing.  If you are up for that, please feel free to go ahead and do it and open a pull request.

surfertas, 2020-08-31:
@clalancette Yes, thanks will work on this and open a PR once complete. Thanks.

clalancette, 2020-09-14:
Yep, I forgot to close it, thanks.

Just as an FYI, if you put something like "fixes #275" in the description of your PRs, the associated issue will get automatically closed when the PR is merged.  Not a huge deal, but makes it so we don't forget things like this :).

closed 2020-09-14 by clalancette

surfertas, 2020-09-14:
Got it thanks a lot for the information.