ros2/geometry2 #285: PyKDL ModuleNotFoundError: No module named 'PyKDL' by importing import tf2_geometry_msgs

michael1309, 2020-07-18:
## Bug report

**Required Info:**

- Operating System:
  - Windows 10
- Installation type:
  - ROS2 Foxy Release
- Version or commit hash:
  - Official Build
- DDS implementation:
  - Default Setup
- Client library (if applicable):
  - 

#### Steps to reproduce issue
I tried to import import tf2_geometry_msgs in a python script.
``` 
import tf2_geometry_msgs
```

The software chrashes with the following error message:
```
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 783, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "C:\dev\ros2_foxy_release\Lib\site-packages\tf2_geometry_msgs\__init__.py", line 1, in <module>
    from .tf2_geometry_msgs import *
  File "C:\dev\ros2_foxy_release\Lib\site-packages\tf2_geometry_msgs\tf2_geometry_msgs.py", line 31, in <module>
    import PyKDL
ModuleNotFoundError: No module named 'PyKDL'
```

#### Expected behavior
No Crash

#### Actual behavior
Crash

sloretz, 2020-08-06:
It looks like the `python_orocos_kdl` package has not been ported to ROS 2 yet. It has an `AMENT_IGNORE` file in the package that will prevent it from being built.

https://github.com/ros2/orocos_kinematics_dynamics/blob/foxy/python_orocos_kdl/AMENT_IGNORE

The python part of `tf2_geometry_msgs` has also not been ported.

https://github.com/ros2/geometry2/blob/7b0f16f4246d7fc5d53668e7f479aa1d2df298f6/tf2_geometry_msgs/package.xml#L20-L24

https://github.com/ros2/geometry2/blob/37204196fc1c23a082b81cbe8edc78290a5d7fbf/tf2_geometry_msgs/CMakeLists.txt#L24-L27

I think the bug here is that `tf2_geometry_msgs` was importable at all: https://github.com/ros2/geometry2/blob/7b0f16f4246d7fc5d53668e7f479aa1d2df298f6/tf2_geometry_msgs/CMakeLists.txt#L21-L22

Beyond that, I think this is a duplicate of https://github.com/ros2/geometry2/issues/110 . If you would be willing to contribute I would start by removing the AMENT_IGNORE from `python_orocos_kdl` and seeing what it needs to build, maybe starting by trying to revive https://github.com/ros2/orocos_kinematics_dynamics/pull/4

closed 2020-08-06 by sloretz

himty, 2021-01-29:
Leaving this here: I deactivated conda by doing `conda deactivate` until even (base) disappeared from terminal. Something with conda was messing with my ros and pykdl installation