ros2/geometry2 #295: static_transform_publisher more intuitive command-line arguments

clalancette, 2020-08-04:
## Bug report

The order of command-line arguments in the `static_transform_publisher` program is hard to remember and follow.  This led directly to the problem described in https://github.com/ros2/geometry2/issues/292.  It would be better if the command-line arguments were much more specific so there is no ambiguity.  We discussed this in https://github.com/ros2/geometry2/issues/292#issuecomment-668054758 and https://github.com/ros2/geometry2/issues/292#issuecomment-668732629 , and the consensus was to add the following command-line arguments:

Group 1:
* `--roll` - specify the roll in radians
* `--pitch` - specify the pitch in radians
* `--yaw` - specify the yaw in radians

Group 2:
* `--roll_deg` - specify the roll in degrees
* `--pitch_deg` - specify the pitch in degrees
* `--yaw_deg` - specify the pitch in degrees

Group 3:
* `--qx` - Specify the quaternion X
* `--qy` - Specify the quaternion Y
* `--qz` - Specify the quaternion Z
* `--qw` - Specify the quaternion W

Note that each of the groups is mutually exclusive; if the user specifies `--roll`, they cannot then specify `--roll_deg` or `--qx` (while we technically could support mixed `--roll` and `--pitch_deg`, we'll just specify that they are exclusive for simplicity).

In order to "tick-tock" this, we'll also continue to support the old command-line options up to and including Galactic, though with a deprecation warning.  After Galactic, we'll remove the old command-line options.  The old command-line options should also be mututally exclusive with any of the 3 groups above.

@allenh1 has graciously offered to take on implementing this.  @Stapelzeiger FYI.

allenh1, 2020-10-10:
@clalancette Starting on this now, but I'm wondering how you want the translation handled? I feel like it's appropriate to add `--x [value] --y [value] --z [value]` (certainly makes the implementation easier, since I can assume things are in pairs), but I also feel like it's kinda silly to do so.

Would you prefer to have a `--translation x y z` or `--xyz x y z`?

Also should `--frame-id frame` and `--child-frame-id child_frame` follow the same pattern? I don't really have any strong opinions here.

clalancette, 2020-10-11:
> @clalancette Starting on this now, but I'm wondering how you want the translation handled? I feel like it's appropriate to add `--x [value] --y [value] --z [value]` (certainly makes the implementation easier, since I can assume things are in pairs), but I also feel like it's kinda silly to do so.

I think we should make translation follow the same pattern as the rotation for consistency.  Thus, I would go with `--x`, `--y`, `--z` here.

> Also should `--frame-id frame` and `--child-frame-id child_frame` follow the same pattern? I don't really have any strong opinions here.

Yeah, I think so.

---
There's a couple of things I failed to think about in the original description.

1.  What happens if the user specifies *no* rotation or *no* translation?  In that case, I think we should assume the identity transformation.
1.  What happens in the user specifies `--x`, but not `--y` or `--z`?  We could either throw an error (meaning that if you specify one of a "group", you have to specify it all), or we could assume identity, but replace the axes that are specified with the user supplied ones.  I'm on the fence with that one.  I slightly lean towards throwing an error, but I could easily be persuaded otherwise.

allenh1, 2020-10-12:
> We could either throw an error (meaning that if you specify one of a "group", you have to specify it all), or we could assume identity, but replace the axes that are specified with the user supplied [...]

I would expect things to work with just `--x [value]` and no `--y` or `--z` specified, but I see where you're coming from. Maybe a warning should be printed?

clalancette, 2020-10-12:
> I would expect things to work with just `--x [value]` and no `--y` or `--z` specified, but I see where you're coming from.

Yeah, that's reasonable.  Since this is going to be pretty verbose, I think we want to allow people to specify the minimum.  I'm just wary of surprises, but in this case I think people will generally assume identity plus whatever they specified.

> Maybe a warning should be printed?

I wouldn't suggest a warning, but maybe something informational.  Like:

```
Applying translation of 0, 0, 0
Applying rotation of 0, 0, 0, 1
```

(you can probably come up with some better wording than that)

allenh1, 2021-01-30:
@clalancette I'll be starting on this in the next day or so. Sorry for the delays.